78/xiaozhi-esp32 #2: How to change the settings for xiao esp32s3 sense

mqgw, 2024-09-14:
Recently I tried to reuse the code for xiao esp32s3 sense. Have tried to change the GPIO as below, but failed to record the voice. Besides these changes, what else needs the change?

    chan_cfg.id = I2S_NUM_1;
    ESP_ERROR_CHECK(i2s_new_channel(&chan_cfg, nullptr, &rx_handle_));
    std_cfg.clk_cfg.sample_rate_hz = (uint32_t)input_sample_rate_;
    std_cfg.gpio_cfg.bclk = (gpio_num_t)42; //CONFIG_AUDIO_DEVICE_I2S_MIC_GPIO_BCLK;
    std_cfg.gpio_cfg.ws = I2S_GPIO_UNUSED;//(gpio_num_t)CONFIG_AUDIO_DEVICE_I2S_MIC_GPIO_WS;
    std_cfg.gpio_cfg.dout = I2S_GPIO_UNUSED;
    std_cfg.gpio_cfg.din = (gpio_num_t)41; //CONFIG_AUDIO_DEVICE_I2S_GPIO_DIN;

78, 2024-09-15:
What do you mean by "wake the chips"?
I am not very familiar with I2S either. I have only been working with it for a month. But as far as I know, the WS pin needs to be set. If you are also using INMP441.
XIAO esp32s3 has too few GPIO pins. You should use duplex mode to share the BCLK and WS pins if you have a speaker and a microphone both connected.

mqgw, 2024-09-22:
Thanks. After several tests, the issue can be solved by changing I2S from std to pdm.

78, 2024-10-25:
I have done it with one I2S.

[image]

closed 2024-11-06 by 78

JosenJin, 2024-12-21:
> Thanks. After several tests, the issue can be solved by changing I2S from std to pdm.

hi friend, I also use xiao esp32 sense, can you share the code? Is your camera integrated?