78/xiaozhi-esp32 #47: managed_components 里的代码编译报错

wi-cuckoo, 2025-01-08:
编译报错，看着是依赖包问题，这块经验不足不知道是不是版本lock不对
环境信息：
* sdk 5.5.0
* board：esp32s3 via usb-jtag
* 代码版本：v0.9.8

报错日志如下
```c
In file included from /workspace/espcar/xiaozhi-esp32/managed_components/78__esp-opus-encoder/opus_decoder.cc:1:
/workspace/espcar/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:29: error: 'uint8_t' was not declared in this scope
   15 |     bool Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm);
      |                             ^~~~~~~
/workspace/espcar/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:8:1: note: 'uint8_t' is defined in header '<cstdint>'; this is probably fixable by adding '#include <cstdint>'
    7 | #include "opus.h"
  +++ |+#include <cstdint>
    8 | 
/workspace/espcar/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:36: error: template argument 1 is invalid
   15 |     bool Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm);
      |                                    ^
/workspace/espcar/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:36: error: template argument 2 is invalid
/workspace/espcar/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:37: error: expected identifier before '&&' token
   15 |     bool Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm);
      |                                     ^~
/workspace/espcar/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:58: error: 'int16_t' was not declared in this scope
   15 |     bool Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm);
      |                                                          ^~~~~~~
/workspace/espcar/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:58: note: 'int16_t' is defined in header '<cstdint>'; this is probably fixable by adding '#include <cstdint>'
/workspace/espcar/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:65: error: template argument 1 is invalid
   15 |     bool Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm);
      |                                                                 ^
/workspace/espcar/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:65: error: template argument 2 is invalid
/workspace/espcar/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:66: error: expected identifier before '&' token
   15 |     bool Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm);
      |                                                                  ^
/workspace/espcar/xiaozhi-esp32/managed_components/78__esp-opus-encoder/opus_decoder.cc:23:6: error: no declaration matches 'bool OpusDecoderWrapper::Decode(std::vector<unsigned char>&&, std::vector<short int>&)'
   23 | bool OpusDecoderWrapper::Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm) {
      |      ^~~~~~~~~~~~~~~~~~
/workspace/espcar/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:10: note: candidate is: 'bool OpusDecoderWrapper::Decode(int&&, int&)'
   15 |     bool Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm);
      |          ^~~~~~
/workspace/espcar/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:10:7: note: 'class OpusDecoderWrapper' defined here
   10 | class OpusDecoderWrapper {
      |       ^~~~~~~~~~~~~~~~~~
[1682/1787] Building CXX object esp-idf/78__esp-ml307/CMakeFiles/__idf_78__esp-ml307.dir/ml307_http.cc.obj
ninja: build stopped: subcommand failed.
```

wi-cuckoo, 2025-01-09:
我按官方教程装的环境，没有注意版本这块，我先试试改下吧

709397578, 2025-01-14:
Activating ESP-IDF 5.4
我的也报错，编译不成功啊

bluejazzCHN, 2025-01-15:
The problem is in managed components. if using esp-idf sdk 5.4 or above, please add #include "stdint.h" into opus_decoder.h find in the folder of managed_components 

build again. all will work.

Simon1413, 2025-01-16:
最新版本的idf不支持？？？？

zhangdiannao, 2025-01-16:
> The problem is in managed components. if using esp-idf sdk 5.4 or above, please add #include "stdint.h" into opus_decoder.h find in the folder of managed_components
> 
> build again. all will work.

Thank you,[bluejazzCHN]!It works.I sucessed build in this way and my sdk is ESP-IDF v5.4.0.Target is ESP32S3.

closed 2025-02-08 by 78