78/xiaozhi-esp32 #48: QSPI screen support?

Y1hsiaochunnn, 2025-01-09:
I have a Waveshare development board ESP32-S3-Touch-AMOLED-1.8, which has ES8311 and power amplifier chip onboard. I have successfully connected to xiaozhi.me. When I enable QSPI, I refer to ESP-IDF esp_lcd driver to write and compile successfully. It does not display the screen after burning.

Maybe it's a code issue?

You can check out the definition here: [ESP32-S3-Touch-AMOLED-1.8](https://github.com/waveshareteam/ESP32-AIChats/blob/4aef7f4b24b5156b9de15b5f47f75c197dde3f53/others/xiaozhi-esp32/main/boards/esp32-s3-touch-amoled-1.8/esp32-s3-touch-amoled-1.8.cc#L120)

ZhouKes, 2025-01-09:
I noticed that your buscfg.data1_io_num and ioconfig.cs_gpio_num are the same pin, maybe that's the reason. In addition, can you open the console to see the output log after the device starts?

Y1hsiaochunnn, 2025-01-09:
Hi @ZhouKes ,
Thank you for your reply. I have rechecked the code and it is indeed the case. However, after I fixed the cs pin, there is still no display, no error output from the terminal, everything should work fine, but the screen is not displayed

Y1hsiaochunnn, 2025-01-09:
Hi @Forairaaaaa ,
Maybe you have a better suggestion? Your Monica project seems to use the same screen

78, 2025-01-09:
Is there a sample piece of initialization code for this screen?

Y1hsiaochunnn, 2025-01-10:
Hi @78 .
Of course, I used the esp_lcd component and placed it in the components directory with test_app inside and I was able to light up the screen normally

here is the start code : [esp_lcd_sh8601 source code](https://github.com/waveshareteam/ESP32-AIChats/blob/3dce8656b4c34acd1b7b5f9a8b11b77409110996/others/xiaozhi-esp32/components/esp_lcd_sh8601/test_apps/main/test_esp_lcd_sh8601.c#L183)

Y1hsiaochunnn, 2025-01-13:
The issue has been resolved and this issue is closed. Can I submit board support later?

closed 2025-01-13 by Y1hsiaochunnn

78, 2025-01-13:
Yes, please!!!
I'm wait for a pull request. My waveshare device is still on the way.

Y1hsiaochunnn, 2025-01-14:
Hi @78 ,
Tested PR,https://github.com/78/xiaozhi-esp32/pull/63

The screens are currently out of stock, so shipment may be late, but don't worry about a resupply soon