78/xiaozhi-esp32 #58: 服务端代码有开源么？

linqingping, 2025-01-13:
服务端代码是否有开源？好像只有ars和tts两个服务其他的项目代码好像没有开源？

lanhung, 2025-01-13:
@78 看看是否有可能把服务器端的也开源，具体包括两部分（1）LLM部分如果要连接到我们自己的，比如私有部署的Qwen2.5或者豆包，以及（2）LLM部分和https://github.com/78/xiaozhi整合起来的方法。或者未来有收费的计划，付费开源的方案，麻烦有可能也告知一下,严格来说现在只是部分开源。

cheetahsec, 2025-01-13:
@78 Looking forward to the server-side open source, so that more power can be brought in to enrich and expand this project

78, 2025-01-13:
目前涉及数据库（与 xiaozhi.me 控制面板有关）的部分代码均未开源部分，安全起见，未来这部分代码也不会开源。会考虑抽时间写一个简单的 demo 提供参考。

Websocket server 程序与 LLM 调用代码推荐自行实现，已有群友完成开发。

neil0306, 2025-01-16:
> 群友完成开发的部分求指路

同求, 非常感谢, 我的star已经准备好了!

DYH89, 2025-01-26:
同求, +1111

kaiyuanwansui, 2025-02-04:
> > 群友完成开发的部分求指路
> 
> 同求, 我的star已经准备好了!

这就是群友的开源：

https://github.com/xinnan-tech/xiaozhi-esp32-server

koth, 2025-02-04:
c++的实现简单参考版，目前还不够流畅，能跑通
https://github.com/koth/ainovel

miaoyizi, 2025-02-28:
> 目前涉及数据库（与 xiaozhi.me 控制面板有关）的部分代码均未开源部分，安全起见，未来这部分代码也不会开源。会考虑抽时间写一个简单的 demo 提供参考。
> 
> Websocket server 程序与 LLM 调用代码推荐自行实现，已有群友完成开发。

希望服务端可以开启一部分功能收费 ，一部分面向学者的免费接入。 收费以保障收费用户的接入质量和促进平台良性发展，免费部分可以提供基础的聊天服务以吸引新用户进来。

biubiubiu-max, 2025-03-19:
插眼

closed 2025-03-21 by 78