78/xiaozhi-esp32 #64: 编译报错，这是啥原因啊

709397578, 2025-01-14:
Running ninja in directory /home/tao/esp32/xiaozhi-esp32/build
Executing "ninja all"...
[1/1] cd /home/tao/esp32/xiaozhi-esp32/build/bootloader/esp-idf/esptool_...loader 0x0 /home/tao/esp32/xiaozhi-esp32/build/bootloader/bootloader.bi
Bootloader binary size 0x4040 bytes. 0x3fc0 bytes (50%) free.
[95/591] Building CXX object esp-idf/78__esp-opus-encoder/CMakeFiles/__idf_78__esp-opus-encoder.dir/opus_decoder.cc.obj
FAILED: esp-idf/78__esp-opus-encoder/CMakeFiles/__idf_78__esp-opus-encoder.dir/opus_decoder.cc.obj
/home/tao/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ -DESP_PLATFORM -DIDF_VER=\"v5.4\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/tao/esp32/xiaozhi-esp32/build/config -I/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include -I/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder -I/home/tao/esp32/esp-idf/components/newlib/platform_include -I/home/tao/esp32/esp-idf/components/freertos/config/include -I/home/tao/esp32/esp-idf/components/freertos/config/include/freertos -I/home/tao/esp32/esp-idf/components/freertos/config/xtensa/include -I/home/tao/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/tao/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/tao/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/home/tao/esp32/esp-idf/components/freertos/esp_additions/include -I/home/tao/esp32/esp-idf/components/esp_hw_support/include -I/home/tao/esp32/esp-idf/components/esp_hw_support/include/soc -I/home/tao/esp32/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/home/tao/esp32/esp-idf/components/esp_hw_support/dma/include -I/home/tao/esp32/esp-idf/components/esp_hw_support/ldo/include -I/home/tao/esp32/esp-idf/components/esp_hw_support/debug_probe/include -I/home/tao/esp32/esp-idf/components/esp_hw_support/port/esp32s3/. -I/home/tao/esp32/esp-idf/components/esp_hw_support/port/esp32s3/include -I/home/tao/esp32/esp-idf/components/heap/include -I/home/tao/esp32/esp-idf/components/heap/tlsf -I/home/tao/esp32/esp-idf/components/log/include -I/home/tao/esp32/esp-idf/components/soc/include -I/home/tao/esp32/esp-idf/components/soc/esp32s3 -I/home/tao/esp32/esp-idf/components/soc/esp32s3/include -I/home/tao/esp32/esp-idf/components/soc/esp32s3/register -I/home/tao/esp32/esp-idf/components/hal/platform_port/include -I/home/tao/esp32/esp-idf/components/hal/esp32s3/include -I/home/tao/esp32/esp-idf/components/hal/include -I/home/tao/esp32/esp-idf/components/esp_rom/include -I/home/tao/esp32/esp-idf/components/esp_rom/esp32s3/include -I/home/tao/esp32/esp-idf/components/esp_rom/esp32s3/include/esp32s3 -I/home/tao/esp32/esp-idf/components/esp_rom/esp32s3 -I/home/tao/esp32/esp-idf/components/esp_common/include -I/home/tao/esp32/esp-idf/components/esp_system/include -I/home/tao/esp32/esp-idf/components/esp_system/port/soc -I/home/tao/esp32/esp-idf/components/esp_system/port/include/private -I/home/tao/esp32/esp-idf/components/xtensa/esp32s3/include -I/home/tao/esp32/esp-idf/components/xtensa/include -I/home/tao/esp32/esp-idf/components/xtensa/deprecated_include -I/home/tao/esp32/esp-idf/components/esp_timer/include -I/home/tao/esp32/esp-idf/components/lwip/include -I/home/tao/esp32/esp-idf/components/lwip/include/apps -I/home/tao/esp32/esp-idf/components/lwip/include/apps/sntp -I/home/tao/esp32/esp-idf/components/lwip/lwip/src/include -I/home/tao/esp32/esp-idf/components/lwip/port/include -I/home/tao/esp32/esp-idf/components/lwip/port/freertos/include -I/home/tao/esp32/esp-idf/components/lwip/port/esp32xx/include -I/home/tao/esp32/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/tao/esp32/esp-idf/components/lwip/port/esp32xx/include/sys -I/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus/include -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -fdiagnostics-color=always -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-error=extra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -mdisable-hardware-atomics -Og -fno-shrink-wrap -fmacro-prefix-map=/home/tao/esp32/xiaozhi-esp32=. -fmacro-prefix-map=/home/tao/esp32/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu++2b -fexceptions -fno-rtti -fuse-cxa-atexit -MD -MT esp-idf/78__esp-opus-encoder/CMakeFiles/__idf_78__esp-opus-encoder.dir/opus_decoder.cc.obj -MF esp-idf/78__esp-opus-encoder/CMakeFiles/__idf_78__esp-opus-encoder.dir/opus_decoder.cc.obj.d -o esp-idf/78__esp-opus-encoder/CMakeFiles/__idf_78__esp-opus-encoder.dir/opus_decoder.cc.obj -c /home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/opus_decoder.cc
In file included from /home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/opus_decoder.cc:1:
/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:29: error: 'uint8_t' was not declared in this scope
   15 |     bool Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm);
      |                             ^~~~~~~
/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:8:1: note: 'uint8_t' is defined in header '<cstdint>'; this is probably fixable by adding '#include <cstdint>'
    7 | #include "opus.h"
  +++ |+#include <cstdint>
    8 |
/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:36: error: template argument 1 is invalid
   15 |     bool Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm);
      |                                    ^
/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:36: error: template argument 2 is invalid
/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:37: error: expected identifier before '&&' token
   15 |     bool Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm);
      |                                     ^~
/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:58: error: 'int16_t' was not declared in this scope
   15 |     bool Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm);
      |                                                          ^~~~~~~
/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:58: note: 'int16_t' is defined in header '<cstdint '; this is probably fixable by adding '#include <cstdint>'
/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:65: error: template argument 1 is invalid
   15 |     bool Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm);
      |                                                                 ^
/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:65: error: template argument 2 is invalid
/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:66: error: expected identifier before '&' token
   15 |     bool Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm);
      |                                                                  ^
/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/opus_decoder.cc:23:6: error: no declaration matches 'bool OpusDecoderWrapper::Decode(std::vector<unsigned char>&&, std::vector<short int>&)'
   23 | bool OpusDecoderWrapper::Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm) {
      |      ^~~~~~~~~~~~~~~~~~
/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:15:10: note: candidate is: 'bool OpusDecoderWrapper::Decode(int&&, int&)'
   15 |     bool Decode(std::vector<uint8_t>&& opus, std::vector<int16_t>& pcm);
      |          ^~~~~~
/home/tao/esp32/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include/opus_decoder.h:10:7: note: 'class OpusDecoderWrapper' defined here
   10 | class OpusDecoderWrapper {
      |       ^~~~~~~~~~~~~~~~~~
[102/591] Building CXX object esp-idf/espressif__esp-dsp/CMakeFiles/__idf_espressif__esp-dsp.dir/modules/matrix/mat/mat.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/tao/esp32/xiaozhi-esp32/build/log/idf_py_stderr_output_171276 and /home/tao/esp32/xiaozhi-esp32/build/log/idf_py_stdout_output_171276

ymthmx, 2025-01-15:
我也是这个报错

78, 2025-01-15:
建议先使用 IDF 5.3 编译。

78, 2025-01-15:
Same question.

https://github.com/78/xiaozhi-esp32/issues/47#issuecomment-2592158027

closed 2025-01-15 by 78

ymthmx, 2025-01-16:
> 建议先使用 IDF 5.3 编译。
[image]