78/xiaozhi-esp32 #66: 疑似waveshare/esp_lcd_sh8601依赖导致编译报错

xiaohuangtx, 2025-01-15:
环境：
- SDK：5.3.2
- board：esp32c3 （虾哥 Mini C3）
- 代码: 最新的main分支代码（bdbbff302d96aa7bb0b99f801969f64bcc71d415）

执行`idf.py set-target esp32c3`报错：
```
PS D:\ESP\test\xiaozhi-esp32> idf.py set-target esp32c3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'D:\ESP\test\xiaozhi-esp32\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32c3, new sdkconfig will be created.
Running cmake in directory D:\ESP\test\xiaozhi-esp32\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Users\ooo\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32c3 -DCCACHE_ENABLE=1 D:\ESP\test\xiaozhi-esp32"...
-- Found Git: C:/Users/ooo/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/ooo/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/ooo/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/ooo/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c3
NOTICE: Dependencies lock doesn't exist, solving dependencies.
.........................HINT: Failed to solve dependencies. Here are some possible reasons:
Component "waveshare/esp_lcd_sh8601" (requires in D:\ESP\test\xiaozhi-esp32\main\idf_component.yml) has suitable versions for other targets:
- 1.0.1: esp32s3
Is your current target esp32c3 set correctly?
HINT: Please check manifest file of the following component(s): main
CMake Error at C:/Users/ooo/esp/v5.3.2/esp-idf/tools/cmake/build.cmake:552 (message):
  ERROR: Because project depends on waveshare/esp_lcd_sh8601 (1.0.1) which
  doesn't match any versions, version solving failed.

Call Stack (most recent call first):
  C:/Users/ooo/esp/v5.3.2/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:10 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the D:\ESP\test\xiaozhi-esp32\build\log\idf_py_stderr_output_16056 and D:\ESP\test\xiaozhi-esp32\build\log\idf_py_stdout_output_16056
PS D:\ESP\test\xiaozhi-esp32> 
```

从报错信息看疑似`waveshare/esp_lcd_sh8601`依赖导致，回退到 `git reset --hard a9244470ef649a8c5f03f76115cb4f79d11578f4` 编译通过。

所以可能是 #63 这会PR没处理好？

78, 2025-01-15:
麻烦提供一下编译环境，比如IDF版本？

Simon1413, 2025-01-16:
[image]

Y1hsiaochunnn, 2025-01-16:
Hi @Simon1413 ,
This problem was caused by the online component library, which has now removed the chip limit
[image]

Hi @78 ,
Submitted a new PR, modify idf_component. Yml the components in the version of https://github.com/78/xiaozhi-esp32/pull/73

xiaohuangtx, 2025-01-16:
> 麻烦提供一下编译环境，比如IDF版本？

@78  上面有提到环境信息,这里补充一下：
- IDF：5.3.2
- OS：Windows 11 24H2 家庭中文版

IDF从乐鑫官网下载的离线安装包`esp-idf-tools-setup-offline-5.3.2.exe`,使用PowerShell终端。

完整信息：
<details><pre><code>
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf5.3_py3.11_env\Scripts
Python 3.11.2
Using Git in C:\Espressif\tools\idf-git\2.44.0\cmd
git version 2.44.0.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v5.3.2
Checking Python compatibility
Adding ESP-IDF tools to PATH...
WARNING: The following issue occurred while accessing the ESP-IDF version file in the Python environment: [Errno 2] No such file or directory: 'C:\\Espressif\\python_env\\idf5.3_py3.11_env\\idf_version.txt'. (Diagnostic information. It can be ignored.)

Name                           Value
----                           -----
OPENOCD_SCRIPTS                C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\openocd\s...
IDF_CCACHE_ENABLE              1
ESP_ROM_ELF_DIR                C:\Espressif\tools\esp-rom-elfs\20240305\
IDF_PYTHON_ENV_PATH            C:\Espressif\python_env\idf5.3_py3.11_env
ESP_IDF_VERSION                5.3
IDF_DEACTIVATE_FILE_PATH       C:\Users\ooo\AppData\Local\Temp\tmphd_cx_ghidf_25740

Added to PATH
-------------
C:\Espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin
C:\Espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin
C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin
C:\Espressif\tools\esp-clang\16.0.1-fe4f10a809\esp-clang\bin
C:\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin
C:\Espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin
C:\Espressif\tools\cmake\3.30.2\bin
C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\bin
C:\Espressif\tools\ninja\1.12.1\
C:\Espressif\tools\idf-exe\1.0.3\
C:\Espressif\tools\ccache\4.10.2\ccache-4.10.2-windows-x86_64
C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64
C:\Espressif\frameworks\esp-idf-v5.3.2\tools
Checking if Python packages are up to date...
Constraint file: C:\Espressif\espidf.constraints.v5.3.txt
Requirement files:
 - C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe
Python requirements are satisfied.

Detected installed tools that are not currently used by active ESP-IDF version.
For removing old versions of amazon-corretto-11-x64-windows-jdk, ccache, cmake, espressif-ide, idf-driver, idf-python-wheels, ninja, openocd-esp32 use command 'python.exe C:\Espressif\frameworks\esp-idf-v5.3.2\tools\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe C:\Espressif\frameworks\esp-idf-v5.3.2\tools\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:
    idf.py build

PS C:\Espressif\frameworks\esp-idf-v5.3.2>

</code></pre></details>

正如 @Y1hsiaochunnn  提的 #73 ，目前上游waveshare/esp_lcd_sh8601解除了限制，将waveshare/esp_lcd_sh8601版本改为1.0.2后编译通过。

我将会在 #73 合并后关闭此issues

Y1hsiaochunnn, 2025-01-16:
great

closed 2025-01-17 by xiaohuangtx