78/xiaozhi-esp32 #76: 希望在后可以选择粤语TTS

neil0306, 2025-01-16:
首先非常感谢虾哥开源了这么棒的repo!

我注意到README中有提到TTS是支持粤语的, 但是我在后台设置里面只找到了"广州德哥(中文)" 这么一个最贴近的角色, 但是他其实是普通话的TTS, 于是, 想了解一下是否可以支持粤语的音源切换呢? 

此外, 我看到另一个xiaozhi仓库里开源了TTS-server的一些代码, 不过我目前简单看了一下似乎并不是能直接部署并用来切换TTS的, 想了解一下如果近期xiaozhi.me的后台切换粤语功能没有更新计划的话, 我是不是有方法可以自行部署测试呢?

78, 2025-01-17:
ASR supports cantonese, but TTS not.
Currently TTS server works with commercial API. I hope they add some cantonese voices too.

kaiyuanwansui, 2025-02-04:
> 首先非常感谢虾哥开源了这么棒的repo!
> 
> 我注意到README中有提到TTS是支持粤语的, 但是我在后台设置里面只找到了"广州德哥(中文)" 这么一个最贴近的角色, 但是他其实是普通话的TTS, 于是, 想了解一下是否可以支持粤语的音源切换呢?
> 
> 此外, 我看到另一个xiaozhi仓库里开源了TTS-server的一些代码, 不过我目前简单看了一下似乎并不是能直接部署并用来切换TTS的, 想了解一下如果近期xiaozhi.me的后台切换粤语功能没有更新计划的话, 我是不是有方法可以自行部署测试呢?

This project can deploy its own backend service and reply in Cantonese：[xiaozhi-esp32-server](https://github.com/xinnan-tech/xiaozhi-esp32-server)

If you are using `EdgeTTS`， `EdgeTTS` sets its voice to `zh-HK-HiuGaaiNeural`、`zh-HK-HiuMaanNeural`、`zh-HK-WanLungNeural ` can reply in Cantonese.

And even if you speak Cantonese, he can understand it because ASR uses `FunASR`。

78, 2025-03-19:
> 如果能支持粤语就好了，可以练习粤语口语。

正在研发阶段，预计最快下个月