78/xiaozhi-esp32 #84: 能否使用ESP32-C3-WROOM-02-H4替换芯片制作xmini-c3版本？

molun, 2025-01-19:
疑问一、ESP32-C3-WROOM-02-H4模组本身只有4M Flash，4M能否满足使用要求？或者开盖更换为16M Flash？
另外就是功放芯片使能引脚修改为模组上的IO1或者IO0是否可以，只许修改代码中13替换为1或0？

[image]

[image]
疑问二、
[image]
另外一个就是虾哥用的这个flash和模组中封装的是否有区别，4线 6线？

[image]
疑问三、
虾哥使用的是串口1，模组中的串口是0，代码中是否需要进行修改，具体位置在哪里？

78, 2025-01-19:
1. IIRC, C3H4 uses internal 4M flash, you cannot replace it. I'd like to choose the IO pins based on convenience. You can change the IO pins as you like. Read the chip manual before you do that to make sure that pins are available.
2. I used 4-lines flash to spare two IO pins for user usage. 6-lines flash brings more security I think. You'd better ask AI about it.
3. UART0 is often used for debugging. However, I choose to use the USB cable instead of the TTL one for debugging and flashing firmwares.

78, 2025-01-19:
No pinyin input in my Linux, just reply you in English.

closed 2025-02-08 by 78