78/xiaozhi-esp32 #111: 使用M5STACK AtomS3R, 配对WIFI时候搜寻不到WIFI

superkevinyen, 2025-01-25:
我使用esp32 example里面的范例烧到m5stack可以抓到很多SSID. 但是使用目前的source code, 并没有办法顺利找到ssid, 也无法强制连接到某个SSID, 这是为什呢?

I (1287) WifiBoard: Free internal: 140891 minimal internal: 140867
I (11297) WifiBoard: Free internal: 144567 minimal internal: 135835
I (14057) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1, snd_ch_cfg:0x0
I (14067) wifi:station: ba:96:f2:98:4b:cb join, AID=1, bgn, 20
I (14067) WifiConfigurationAp: Station ba:96:f2:98:4b:cb joined, AID=1
I (14197) wifi:<ba-add>idx:2 (ifx:1, ba:96:f2:98:4b:cb), tid:0, ssn:0, winSize:64
I (14197) wifi:<ba-add>idx:3 (ifx:1, ba:96:f2:98:4b:cb), tid:6, ssn:0, winSize:64
I (14227) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (14497) DnsServer: Sending DNS response to 192.168.4.1
I (14497) DnsServer: Sending DNS response to 192.168.4.1
I (14497) DnsServer: Sending DNS response to 192.168.4.1
I (18607) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (21297) WifiBoard: Free internal: 139259 minimal internal: 127383
I (25397) WifiConfigurationAp: SSID: Seraphim, RSSI: -90, Authmode: 3
I (31297) WifiBoard: Free internal: 141391 minimal internal: 126283
I (41297) WifiBoard: Free internal: 141319 minimal internal: 126283

superkevinyen, 2025-01-26:
应该是说 我用你的扫不到AP。这份Log 是我在其他wifi router扫出来的。这附近应该有-65 到-68db 。但自己写了一个比较简单apsta扫router 却可以扫出来10几个。感觉不知道哪里有问题。

superkevinyen, 2025-01-27:
不知道有没有人遇到这个问题或知道如何解决 我用的是idf 5.4的版本。我现在是没有办法顺利扫描到 其他wifi。

HonestQiao, 2025-01-28:
[image]

这样试试，看看会如何

superkevinyen, 2025-02-10:
目前这样的设定, 确实感觉是可以增加兼容性, 不知道是为什么, 是因为TX把RX的信号给覆盖掉了吗? 造成干扰吗?我测试了下, 确实兼容性得到了保证. 打算在menuconfig wifi里面可以新添加一个 "WIFI_BETTER_COMPATIABLE" 来使用.

78, 2025-02-10:
所以这是什么引起的问题呢？

superkevinyen, 2025-02-11:
如果是因为TX把RX造成干扰, 那就是因为可能内部结构天线位置有问题. 我目前使用这样避开
.

LeeRHuang, 2025-04-17:
如果在单位有很多AP，一直连不上，加上这句好使，ESP_ERROR_CHECK(esp_wifi_set_max_tx_power(34));

78, 2025-04-17:
> 如果在单位有很多AP，一直连不上，加上这句好使，ESP_ERROR_CHECK(esp_wifi_set_max_tx_power(34));

它的默认值是多少？

78, 2025-05-20:
配网模式已支持配置tx power。

closed 2025-05-20 by 78