78/xiaozhi-esp32 #117: 厚国兄的音频集成板没有声音

q629988171, 2025-01-28:
print("祝你新年快乐，代码无Bug！")

使用ESP32-S3与厚国兄的音频集成板（MAX98357音频放大器模块）连接，固件使用的是小智AI （v1.0.1），控制台显示已经联网成功并且文本方式输出对话正常，GPIO配置与代码中的设置一致。

厚国兄的音频集成板跑测试代码能输出声音。

```
#include <stdio.h>
#include <math.h>
#include "freertos/freertos.h"
#include "freertos/task.h"
#include "driver/i2s.h"
#include "driver/gpio.h"

// 配置参数
#define SAMPLE_RATE     (44100)
#define I2S_NUM         (I2S_NUM_0)
#define WAVE_FREQ_HZ    (1000)      // 1kHz正弦波
#define AMPLITUDE       (0.5)       // 输出幅度（0.0-1.0）

// 根据新引脚表修改的GPIO定义
#define I2S_BCLK_GPIO   (GPIO_NUM_15)  // 对应BCLK
#define I2S_LRCLK_GPIO  (GPIO_NUM_16)  // 对应LRCLK(WS)
#define I2S_DIN_GPIO    (GPIO_NUM_7)   // 对应DIN
#define SD_MODE_GPIO    (GPIO_NUM_6)   // 模式选择

// 生成正弦波数据（优化内存访问）
static void generate_sine_wave(uint32_t *buffer, size_t samples, double amplitude)
{
    const double delta = (2 * M_PI * WAVE_FREQ_HZ) / SAMPLE_RATE;
    
    for (int i = 0; i < samples; i++) {
        int16_t sample = (int16_t)(sin(i * delta) * amplitude * INT16_MAX);
        buffer[i] = (sample << 16) | (sample & 0xFFFF); // 立体声数据
    }
}

void app_main(void)
{
    // 初始化SD模式引脚
    gpio_reset_pin(SD_MODE_GPIO);
    gpio_set_direction(SD_MODE_GPIO, GPIO_MODE_OUTPUT);
    gpio_set_level(SD_MODE_GPIO, 1);  // 启用I2S模式

    // I2S配置
    i2s_config_t i2s_config = {
        .mode = I2S_MODE_MASTER | I2S_MODE_TX,
        .sample_rate = SAMPLE_RATE,
        .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
        .channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT, // 单声道输出
        .communication_format = I2S_COMM_FORMAT_STAND_I2S,
        .dma_buf_count = 8,
        .dma_buf_len = 512,          // 增大缓冲区减少中断
        .use_apll = true,            // 使用高精度时钟
        .tx_desc_auto_clear = true,  // 自动清除DMA描述符
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1
    };

    // 引脚配置
    i2s_pin_config_t pin_config = {
        .bck_io_num = I2S_BCLK_GPIO,
        .ws_io_num = I2S_LRCLK_GPIO,
        .data_out_num = I2S_DIN_GPIO,
        .data_in_num = I2S_PIN_NO_CHANGE
    };

    // 安装并初始化I2S驱动
    ESP_ERROR_CHECK(i2s_driver_install(I2S_NUM, &i2s_config, 0, NULL));
    ESP_ERROR_CHECK(i2s_set_pin(I2S_NUM, &pin_config));

    // 生成音频缓冲区（优化为动态生成）
    const size_t buffer_size = 1024;
    uint32_t *audio_buffer = heap_caps_malloc(buffer_size * sizeof(uint32_t), MALLOC_CAP_DMA);
    generate_sine_wave(audio_buffer, buffer_size, AMPLITUDE);

    // 循环播放
    size_t bytes_written = 0;
    while (1) {
        i2s_write(I2S_NUM, audio_buffer, buffer_size * sizeof(uint32_t), &bytes_written, portMAX_DELAY);
        
        // 动态频率修改示例（可选）
        // static int freq = 1000;
        // if(freq++ > 2000) freq = 1000;
        // generate_sine_wave(audio_buffer, buffer_size, AMPLITUDE);
    }

    // 清理资源（通常不会执行到这里）
    free(audio_buffer);
    i2s_driver_uninstall(I2S_NUM);
}

```

whtwhtw, 2025-02-09:
在对应开发版的config.h文件中修改为：
#define AUDIO_I2S_MIC_GPIO_WS   GPIO_NUM_5
#define AUDIO_I2S_MIC_GPIO_SCK  GPIO_NUM_4
这2个引脚互换一下，厚国兄的电路和面包板的搞反了

whtwhtw, 2025-02-09:
mic电路不一样，在boards目录对应开发板的config.h中修改MIC_GPIO管脚定义
#define AUDIO_I2S_MIC_GPIO_WS   GPIO_NUM_5
#define AUDIO_I2S_MIC_GPIO_SCK  GPIO_NUM_4
交换4,5管脚后mic可以使用了，目前speak不出声还没找到原因。确定服务器返回对话内容了
I (13816) Application: >> 小智
I (13826) Application: STATE: speaking
I (14456) Application: << 嗨！小智在这儿呢，你咋了？
I (15486) main: Free internal: 27963 minimal internal: 21707
I (16826) Application: << 有啥开心或者不开心的事儿想找我聊聊吗？
I (21506) Application: STATE: listening

q629988171, 2025-02-09:
> mic电路不一样，在boards目录对应开发板的config.h中修改MIC_GPIO管脚定义 #define AUDIO_I2S_MIC_GPIO_WS GPIO_NUM_5 #define AUDIO_I2S_MIC_GPIO_SCK GPIO_NUM_4 交换4,5管脚后mic可以使用了，目前speak不出声还没找到原因。确定服务器返回对话内容了 I (13816) [...]

是的，麦克风工作正常，就是喇叭没有声音。

whtwhtw, 2025-02-09:
确定是配置问题，在https://github.com/domonic18/esp32s3-ai-chat项目中的example/audio_player例程又是可以发声的。。。

whtwhtw, 2025-02-09:
确实解决问题，把厚国兄板子的R1短接后，小智会说话了。。。

78, 2025-02-09:
小智固件默认读取的是左声道的音频数据，有些板子可能是配置了右声道，你可以试试在mask那里把 LEFT 改为 RIGHT。

q629988171, 2025-02-09:
> 小智固件默认读取的是左声道的音频数据，有些板子可能是配置了右声道，你可以试试在mask那里把 LEFT 改为 RIGHT。

改了之后麦克风和喇叭没有用。

main\audio_codecs\no_audio_codec.cc
`.slot_mask = I2S_STD_SLOT_LEFT -> I2S_STD_SLOT_RIGHT`