78/xiaozhi-esp32 #118: esp32-s3无法找到wifi热点

marvelliu, 2025-01-29:
大家好，我买的是esp32-s3-n16r8板子

然后用最新的[v1.0.1_bread-compact-wifi.zip]固件，烧录没有问题，启动的时候也有声音提示，有wifi热点和url提示，但手机和电脑都找不到xiaozhi-的热点。
用超级终端连接，输出如下，请问问题是什么呢？

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x568
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc14
load:0x403cb700,len:0x2d3c
entry 0x403c8904
[0;32mI (36) octal_psram: vendor id    : 0x0d (AP)[0m
[0;32mI (36) octal_psram: dev id       : 0x02 (generation 3)[0m
[0;32mI (36) octal_psram: density      : 0x03 (64 Mbit)[0m
[0;32mI (41) octal_psram: good-die     : 0x01 (Pass)[0m
[0;32mI (46) octal_psram: Latency      : 0x01 (Fixed)[0m
[0;32mI (51) octal_psram: VCC          : 0x01 (3V)[0m
[0;32mI (56) octal_psram: SRF          : 0x01 (Fast Refresh)[0m
[0;32mI (62) octal_psram: BurstType    : 0x01 (Hybrid Wrap)[0m
[0;32mI (68) octal_psram: BurstLen     : 0x01 (32 Byte)[0m
[0;32mI (73) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)[0m
[0;32mI (79) octal_psram: DriveStrength: 0x00 (1/1)[0m
[0;32mI (85) MSPI Timing: PSRAM timing tuning index: 5[0m
[0;32mI (89) esp_psram: Found 8MB PSRAM device[0m
[0;32mI (94) esp_psram: Speed: 80MHz[0m
[0;32mI (98) cpu_start: Multicore app[0m
[0;32mI (113) cpu_start: Pro cpu start user code[0m
[0;32mI (113) cpu_start: cpu freq: 240000000 Hz[0m
[0;32mI (113) app_init: Application information:[0m
[0;32mI (116) app_init: Project name:     xiaozhi[0m
[0;32mI (121) app_init: App version:      1.0.1[0m
[0;32mI (126) app_init: Compile time:     Jan 24 2025 02:39:38[0m
[0;32mI (132) app_init: ELF file SHA256:  16051bdf3...[0m
[0;32mI (137) app_init: ESP-IDF:          v5.3.1-dirty[0m
[0;32mI (142) efuse_init: Min chip rev:     v0.0[0m
[0;32mI (147) efuse_init: Max chip rev:     v0.99 [0m
[0;32mI (152) efuse_init: Chip rev:         v0.2[0m
[0;32mI (157) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (164) heap_init: At 3FCAD018 len 0003C6F8 (241 KiB): RAM[0m
[0;32mI (170) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM[0m
[0;32mI (176) heap_init: At 600FE104 len 00001ED4 (7 KiB): RTCRAM[0m
[0;32mI (183) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator[0m
[0;32mI (191) spi_flash: detected chip: boya[0m
[0;32mI (194) spi_flash: flash io: qio[0m
[0;32mI (199) sleep: Configure to isolate all GPIO pins in sleep state[0m
[0;32mI (205) sleep: Enable automatic switching of GPIO sleep configuration[0m
[0;32mI (213) main_task: Started on CPU0[0m
[0;32mI (243) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;32mI (243) main_task: Calling app_main()[0m
[0;32mI (253) BackgroundTask: background_task started[0m
[0;32mI (253) button: IoT Button Version: 3.5.0[0m
[0;32mI (253) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (263) button: IoT Button Version: 3.5.0[0m
[0;32mI (273) gpio: GPIO[47]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (283) button: IoT Button Version: 3.5.0[0m
[0;32mI (283) gpio: GPIO[40]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (293) button: IoT Button Version: 3.5.0[0m
[0;32mI (293) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (303) gpio: GPIO[41]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (313) gpio: GPIO[42]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (323) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (333) Application: STATE: starting[0m
[0;32mI (343) Ssd1306Display: Initialize LVGL[0m
[0;32mI (343) LVGL: Starting LVGL task[0m
[0;32mI (343) Ssd1306Display: Install SSD1306 driver[0m
[0;32mI (353) Ssd1306Display: SSD1306 driver installed[0m
[0;32mI (363) Ssd1306Display: Turning display on[0m
[0;32mI (463) Ssd1306Display: Adding LCD screen[0m
[0;32mI (463) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (463) NoAudioCodec: Simplex channels created[0m
[0;32mI (473) AudioCodec: Set input enable to true[0m
[0;32mI (473) AudioCodec: Set output enable to true[0m
[0;32mI (483) Application: STATE: configuring[0m
[0;32mI (483) DnsServer: Starting DNS server[0m
[0;32mI (483) pp: pp rom version: e7ae62f[0m
[0;32mI (493) net80211: net80211 rom version: e7ae62f[0m
I (503) wifi:wifi driver task: 3fccc6c0, prio:23, stack:6656, core=0
I (503) wifi:wifi firmware version: ccaebfa
I (503) wifi:wifi certification version: v7.0
I (513) wifi:config NVS flash: enabled
I (513) wifi:config nano formating: disabled
I (523) wifi:Init data frame dynamic rx buffer num: 32
I (523) wifi:Init static rx mgmt buffer num: 5
I (523) wifi:Init management short buffer num: 32
I (533) wifi:Init static tx buffer num: 16
I (533) wifi:Init tx cache buffer num: 32
I (543) wifi:Init static tx FG buffer num: 2
I (543) wifi:Init static rx buffer size: 1600
I (543) wifi:Init static rx buffer num: 16
I (553) wifi:Init dynamic rx buffer num: 32
[0;32mI (553) wifi_init: rx ba win: 16[0m
[0;32mI (563) wifi_init: accept mbox: 6[0m
[0;32mI (563) wifi_init: tcpip mbox: 32[0m
[0;32mI (563) wifi_init: udp mbox: 6[0m
[0;32mI (573) wifi_init: tcp mbox: 6[0m
[0;32mI (573) wifi_init: tcp tx win: 5760[0m
[0;32mI (583) wifi_init: tcp rx win: 5760[0m
[0;32mI (583) wifi_init: tcp mss: 1440[0m
[0;32mI (583) wifi_init: WiFi/LWIP prefer SPIRAM[0m
I (593) wifi:Set ps type: 0, coexist: 0

[0;32mI (593) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10[0m
I (643) wifi:mode : sta (94:a9:90:14:8b:60) + softAP (94:a9:90:14:8b:61)
I (643) wifi:enable tsf
I (643) wifi:Total power save buffer number: 8
I (643) wifi:Init max length of beacon: 752/752
I (643) wifi:Init max length of beacon: 752/752
[0;32mI (653) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1[0m
[0;32mI (653) WifiConfigurationAp: Access Point started with SSID Xiaozhi-148B61[0m
[0;32mI (663) WifiConfigurationAp: Web server started[0m
[0;33mW (673) Application: Alert: Info, ??????????????????[0m
[0;32mI (673) Application: PlayLocalFile: 3038 bytes[0m
[0;32mI (683) Application: Resampling audio from 16000 to 24000[0m
[0;32mI (683) OpusResampler: Resampler configured with input sample rate 16000 and output sample rate 24000[0m
[0;32mI (703) WifiBoard: Free internal: 110903 minimal internal: 110879[0m
[0;32mI (10703) WifiBoard: Free internal: 114771 minimal internal: 106043[0m
[0;32mI (20703) WifiBoard: Free internal: 114771 minimal internal: 106043[0m
[0;32mI (30703) WifiBoard: Free internal: 114771 minimal internal: 106043[0m

marvelliu, 2025-01-29:
偶尔有xiaozhi出现，但点击连接后马上断掉，然后又消失了

marvelliu, 2025-01-29:
我把esp32板子从面包板拔出来，这次能看到热点了，连上填入我的wifi名称和密码，但连接不上，不知为何

I (60723) wifi:mode : sta (94:a9:90:14:8b:60) + softAP (94:a9:90:14:8b:61)
I (60733) wifi:enable tsf
I (60733) wifi:Total power save buffer number: 8
I (60733) wifi:Init max length of beacon: 752/752
I (60743) wifi:Init max length of beacon: 752/752
[0;32mI (60743) WifiConfigurationAp: Access Point started with SSID Xiaozhi-148B61[0m
[0;32mI (60743) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1[0m
[0;32mI (60763) WifiConfigurationAp: Web server started[0m
[0;33mW (60763) Application: Alert: Info, ??????????????????[0m
[0;32mI (60773) Application: PlayLocalFile: 3038 bytes[0m
[0;32mI (60783) Application: Resampling audio from 16000 to 24000[0m
[0;32mI (60783) OpusResampler: Resampler configured with input sample rate 16000 and output sample rate 24000[0m
[0;32mI (60793) WifiBoard: Free internal: 121803 minimal internal: 121779[0m
[0;32mI (70803) WifiBoard: Free internal: 125603 minimal internal: 116943[0m
I (73033) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1, snd_ch_cfg:0x0
I (73033) wifi:station: 52:77:ff:d6:ea:7c join, AID=1, bgn, 40U
[0;32mI (73033) WifiConfigurationAp: Station 52:77:ff:d6:ea:7c joined, AID=1[0m
I (73123) wifi:<ba-add>idx:2 (ifx:1, 52:77:ff:d6:ea:7c), tid:0, ssn:0, winSize:64
[0;32mI (73143) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2[0m
[0;32mI (74573) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (74583) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (74583) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (74583) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (74593) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (74593) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (74603) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;33mW (75793) httpd_uri: httpd_uri: URI '/generate_204_3958c010-bcca-4c7e-9518-0e00035add9d' not found[0m
[0;33mW (75793) httpd_txrx: httpd_resp_send_err: 404 Not Found - Nothing matches the given URI[0m
[0;32mI (76023) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (76973) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (76973) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (77013) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (77013) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (77133) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (77733) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (77843) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (78143) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (78393) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (78393) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (79623) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (79623) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (80023) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (80483) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (80483) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (80483) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (80483) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (80493) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (80703) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (80763) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (80793) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (80793) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (80793) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (80803) WifiBoard: Free internal: 125171 minimal internal: 116943[0m
I (80803) wifi:[0;32mI (80803) DnsServer: Sending DNS response to 192.168.4.1[0m
<ba-add>idx:3 (ifx:1, 52:77:ff:d6:ea:7c), tid:6, ssn:0, winSize:64
[0;32mI (80813) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (80953) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (81013) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (81023) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (81043) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (81133) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (81133) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (81153) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (81203) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (81273) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (82013) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (82023) DnsServer: Sending DNS response to 192.168.4.1[0m
[0;32mI (82053) DnsServer: Sending DNS response to 192.168.4.1[0m
[... 292 lines cut ...]
[0;32mI (60753) wifi_init: tcpip mbox: 32[0m
[0;32mI (60763) wifi_init: udp mbox: 6[0m
[0;32mI (60763) wifi_init: tcp mbox: 6[0m
[0;32mI (60773) wifi_init: tcp tx win: 5760[0m
[0;32mI (60773) wifi_init: tcp rx win: 5760[0m
[0;32mI (60773) wifi_init: tcp mss: 1440[0m
[0;32mI (60783) wifi_init: WiFi/LWIP prefer SPIRAM[0m
I (60793) wifi:Set ps type: 0, coexist: 0

I (60793) wifi:mode : sta (94:a9:90:14:8b:60) + softAP (94:a9:90:14:8b:61)
I (60793) wifi:enable tsf
I (60803) wifi:Total power save buffer number: 8
I (60803) wifi:Init max length of beacon: 752/752
[... cut ...]
[0;32mI (60843) Application: Resampling audio from 16000 to 24000[0m
[0;32mI (60853) OpusResampler: Resampler configured with input sample rate 16000 and output sample rate 24000[0m
[0;32mI (60863) WifiBoard: Free internal: 108427 minimal internal: 108403[0m
[0;32mI (70873) WifiBoard: Free internal: 112251 minimal internal: 104387[0m
[0;32mI (80873) WifiBoard: Free internal: 112251 minimal internal: 104387[0m
[0;32mI (90873) WifiBoard: Free internal: 112251 minimal internal: 104387[0m

marvelliu, 2025-01-29:
不知道是不是电路板的问题，先问问商家

closed 2025-01-29 by marvelliu

lyhistory, 2025-04-13:
天线问题，如果是用面包板，天线最好伸出面包板边缘，否则你可以在天线上缠几圈导线，简单的测试是你用手指摸着天线区域，看看信号是不是稳定