78/xiaozhi-esp32 #119: 可以调用自己部署的模型API 或者其他大模型API吗？

ThrallOtaku, 2025-01-31:
可以调用自己部署的模型API 或者其他大模型API吗？    应该在哪个文件改代码？

WangCY037, 2025-02-01:
同求，希望能增加api调用功能

78, 2025-02-02:
目前正在内测切换大模型的功能，但由于普遍大模型 API 延迟较为严重，会额外增加1～2秒的应答时间，导致体验不佳。如果希望自定义模型 API，可以考虑采用其他开源项目如 ESPAI 的方案。

lanhung, 2025-02-02:
其实也不是所有的客户都需要非常及时的响应，但是很多客户都需要连接到自己微调后的大语言模型上，所以有时候增加五秒都没关系，主要是连接大模型的接口能不能开放一下，或者至少出一个demo连一个大模型？

Zc365, 2025-02-02:
> 目前正在内测切换大模型的功能，但由于普遍大模型 API 延迟较为严重，会额外增加1～2秒的应答时间，导致体验不佳。如果希望自定义模型 API，可以考虑采用其他开源项目如 ESPAI 的方案。

可以后台模型应该改用deepseek吧？应该比qwen强

kaiyuanwansui, 2025-02-02:
https://github.com/xinnan-tech/xiaozhi-esp32-server

有群友实现了简单版本的后端，可以对接dify，自由编排

960208781, 2025-02-10:
> > 目前正在内测切换大模型的功能，但由于普遍大模型 API 延迟较为严重，会额外增加1～2秒的应答时间，导致体验不佳。如果希望自定义模型 API，可以考虑采用其他开源项目如 ESPAI 的方案。
> 
> 后台模型可以改用deepseek吧？应该比qwen强

有提供deepseek的, 毕竟是推理模型, 都会有延迟, 体验不太好

NevermoreN, 2025-02-13:
> 目前正在内测切换大模型的功能，但由于普遍大模型 API 延迟较为严重，会额外增加1～2秒的应答时间，导致体验不佳。如果希望自定义模型 API，可以考虑采用其他开源项目如 ESPAI 的方案

可以不用多快 但是可以给用户自己一个选择嘛

closed 2025-03-21 by 78