78/xiaozhi-esp32 #129: "Have you considered adding Bluetooth functionality to the ESP32 chip?"

dujianmin, 2025-02-06:
The ESP32 chip has classic Bluetooth functionality. After adding Bluetooth features, it can connect to Bluetooth speakers or car audio systems.

dujianmin, 2025-02-06:
因为我看到现在电路板有支持ESP32芯片的电路板了。单独写蓝牙功能正常。代码写入整体代码后。内存溢出。找不到原因

78, 2025-02-08:
update到最新的代码，再没有加蓝牙功能的时候，看看每10秒打印的minimal free internal memory还有多少。或者你试试带psram的芯片

dujianmin, 2025-02-11:
I (17508) main: Free internal: 72635 minimal internal: 42911
I (27508) main: Free internal: 72635 minimal internal: 42911
I (37508) main: Free internal: 72635 minimal internal: 42911
I (47508) main: Free internal: 72635 minimal internal: 42911
I (57508) main: Free internal: 72635 minimal internal: 42911
I (67508) main: Free internal: 72647 minimal internal: 42911

78, 2025-02-11:
看看你说的内存溢出，是否是stack overflow，需要增加任务的堆栈大小

zhuzhe1983, 2025-03-15:
是啊，要是能4G上网和蓝牙车载连接岂不美哉，虾哥考虑支持下？

78, 2025-03-15:
> 是啊，要是能4G上网和蓝牙车载连接岂不美哉，虾哥考虑支持下？

需要外接蓝牙模块，已经有群友实现，可以等他有空出教程

9cat, 2025-03-17:
是ESP32的指标太小。 所以无法直接用它的蓝牙来直接做功放输出 与 耳机？

如果 能实现的话。 这2个模块都不需要了。 不是更便捷？

waltcow, 2025-04-01:
使用nimble stack 控制蓝牙的玩具 
```
At 0x3fcb3648 len 221384 free 1864 allocated 216600 min_free 4
At 0x3fce9710 len 22308 free 4 allocated 21432 min_free 4
```

esp32S3 虽然有 8MB PSRAM，但是 BLE 控制器只能用 内部 SRAM，不能用 PSRAM。而程序在启动时（Wi-Fi + 音频 + OLED + LVGL + AFE + MQTT + 模型加载）几乎把内部 SRAM 占光了。

```bash

 (12716) TrainBLE: Initializing BT controller
I (12716) TrainBLE: Before BLE init:
Heap summary for capabilities 0x00001000:
  At 0x3c2c0000 len 8388608 free 8175168 allocated 210604 min_free 8154224
    largest_free_block 8126464 alloc_blocks 134 free_blocks 9 total_blocks 143
  At 0x3fcb3648 len 221384 free 1864 allocated 216604 min_free 4
    largest_free_block 1152 alloc_blocks 494 free_blocks 7 total_blocks 501
  At 0x3fce9710 len 22308 free 4 allocated 21416 min_free 4
    largest_free_block 0 alloc_blocks 38 free_blocks 0 total_blocks 38
  At 0x600fe11c len 7868 free 132 allocated 7212 min_free 4
    largest_free_block 120 alloc_blocks 36 free_blocks 1 total_blocks 37
  Totals:
    free 8175428 allocated 457572 min_free 8154236 largest_free_block 8126464
I (12766) Application: << 火车已经启动啦，如果要停的话随时告诉我哦。
I (12776) BLE_INIT: BT controller compile version [d752dea]
E (12786) BLE_INIT: Malloc failed
E (12786) BLE_INIT: esp_bt_controller_init -4

E (12786) TrainBLE: Failed to initialize controller: ESP_ERR_NO_MEM
E (12796) TrainBLE: BT controller initialization failed
E (12796) TrainBLE: BLE initialization failed, cannot advertise
```

78, 2025-04-01:
> 使用nimble stack 控制蓝牙的玩具
> 
> ```
> At 0x3fcb3648 len 221384 free 1864 allocated 216600 min_free 4
> At 0x3fce9710 len 22308 free 4 allocated 21432 min_free 4
> ```
> 
> esp32S3 虽然有 8MB PSRAM，但是 BLE 控制器只能用 内部 SRAM，不能用 PSRAM。而程序在启动时（Wi-Fi + 音频 + OLED + LVGL + AFE + MQTT + 模型加载）几乎把内部 SRAM 占光了。
> 
>  (12716) TrainBLE: Initializing BT controller
> I (12716) TrainBLE: Before BLE init:
> Heap summary for capabilities 0x00001000:
>   At 0x3c2c0000 len 8388608 free 8175168 allocated 210604 min_free 8154224
>     largest_free_block 8126464 alloc_blocks 134 free_blocks 9 total_blocks 143
>   At 0x3fcb3648 len 221384 free 1864 allocated 216604 min_free 4
>     largest_free_block 1152 alloc_blocks 494 free_blocks 7 total_blocks 501
>   At 0x3fce9710 len 22308 free 4 allocated 21416 min_free 4
>     largest_free_block 0 alloc_blocks 38 free_blocks 0 total_blocks 38
>   At 0x600fe11c len 7868 free 132 allocated 7212 min_free 4
>     largest_free_block 120 alloc_blocks 36 free_blocks 1 total_blocks 37
>   Totals:
>     free 8175428 allocated 457572 min_free 8154236 largest_free_block 8126464
> I (12766) Application: << 火车已经启动啦，如果要停的话随时告诉我哦。
> I (12776) BLE_INIT: BT controller compile version [d752dea]
> E (12786) BLE_INIT: Malloc failed
> E (12786) BLE_INIT: esp_bt_controller_init -4
> 
> E (12786) TrainBLE: Failed to initialize controller: ESP_ERR_NO_MEM
> E (12796) TrainBLE: BT controller initialization failed
> E (12796) TrainBLE: BLE initialization failed, cannot advertise

考虑在启动小智程序之前，先初始化蓝牙功能试试。不然小智可能为了最大化性能，吃掉了所有SRAM。

78, 2025-04-01:
你可以在wifi初始化之前，打印一下内存信息，看输出是不足了。