78/xiaozhi-esp32 #139: declaration of function 'lv_keyboard_create'; did you mean 'lv_bar_create'?

zhaoyangpp, 2025-02-10:
[image]

HonestQiao, 2025-02-10:
idf.py menuconfig，然后取消如下设置的勾选：

[image]

zhaoyangpp, 2025-02-11:
> idf.py menuconfig，然后取消如下设置的勾选：
> 
> <img alt="Image" width="909" src="https://private-user-images.githubusercontent.com/696708/411553678-f3f25c9f-64da-48b1-b6b5-c4ab03f8c4d0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzkyMzk0NzUsIm5iZiI6MTczOTIzOTE3NSwicGF0aCI6Ii82OTY3MDgvNDExNTUzNjc4LWYzZjI1YzlmLTY0ZGEtNDhiMS1iNmI1LWM0YWIwM2Y4YzRkMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIxMVQwMTU5MzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kZmQ0ZDY1YTA4ZjQ4NTAzOTc0Zjk4ZTUxZjlkNGRmZDkwMDgwMzY1YjJjZmUwN2RjZDRhZTJiYTljZDNkYmZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.7mTkC5E301XfI5QMJfyOiaIB5OdFOB2Z6Wh-kHUhA9Q">

好的谢谢 已解决

closed 2025-02-11 by 78