78/xiaozhi-esp32 #148: 我可以在底层文件中新增函数然后PR上来么，由于硬件设计限制  左右声道有所要求

Sail-211010, 2025-02-12:
[image]
在硬件设计中，左右声道固件，硬件上并不好修改相关配置，可以在程序中新增一个函数，在参数上配置左右声道参数么
，新增的函数和 NoAudioCodecSimplex::NoAudioCodecSimplex(int input_sample_rate, int output_sample_rate, gpio_num_t spk_bclk, gpio_num_t spk_ws, gpio_num_t spk_dout, gpio_num_t mic_sck, gpio_num_t mic_ws, gpio_num_t mic_din) 同级

Sail-211010, 2025-02-12:
喇叭和麦克风分别处于左右声道

78, 2025-02-12:
需要同步各boards下用到的文件，你可以提交一个PR。如果只是你自己使用，建议直接开个分支自己修改即可，暂时不推荐在生产中使用NoAudioCodecSimplex这个。

Sail-211010, 2025-02-13:
暂时不推荐在生产中使用NoAudioCodecSimplex这个。这个是因为什么呢    因为我使用的是两个I2S  目前只看到这个函数   有其他函数可以实现相同功能么

78, 2025-02-13:
成本低的设计可以参考 https://oshwhub.com/tenclass01/xmini_c3 或者 立创实战派开发板

Sail-211010, 2025-02-14:
产品是之前设计的   我后面PR上来  你帮忙看下，修改的地方我都是以最小变动进行修改，费心了哈

78, 2025-02-14:
嗯，可以先开个分支使用，定期合并主线代码

closed 2025-05-20 by 78