78/xiaozhi-esp32 #151: [问题] 4G 版固件无法连接 MQTT broker

ShiinaKaze, 2025-02-13:
[image]

[image]

78, 2025-02-13:
信号不好吗？

alienhsu, 2025-02-14:
我用的是wifi版的，也是出现类似的问题，也是无法连接上MQTT。

[image]

78, 2025-02-14:
目前这个问题收到的反馈不多，看看是否跟本地网络有关系呢？

alienhsu, 2025-02-14:
> 目前这个问题收到的反馈不多，看看是否跟本地网络有关系呢？

这个aliyun的服务是正常的，对吧？

ShiinaKaze, 2025-02-14:
> > 目前这个问题收到的反馈不多，看看是否跟本地网络有关系呢？
> > 图中访问的aliyun的服务是正常的，对吧？

阿里云服务器确实正常。

ShiinaKaze, 2025-02-14:
> > 目前这个问题收到的反馈不多，看看是否跟本地网络有关系呢？
> > 图中访问的aliyun的服务是正常的，对吧？

但我这边用ML307连不上

[image]

78, 2025-02-14:
访问其他网站可以用吗？
物联网卡在某些地区确实不太好用。

ShiinaKaze, 2025-02-14:
> 访问其他网站可以用吗？ 物联网卡在某些地区确实不太好用。

已解决，硬件问题，ML307模块有问题。

closed 2025-02-14 by ShiinaKaze