78/xiaozhi-esp32 #164: 【征集讨论】固件生成UUID与添加设备逻辑前置

flysnoworg, 2025-02-14:
### Discussed in https://github.com/78/xiaozhi-esp32/discussions/162

<div type='discussions-op-text'>

<sup>Originally posted by **78** February 14, 2025</sup>
实际上，此为两个事项。

## 固件生成 UUID
自开源项目发布以来，固件生成 UUID 一直以 ESP32 芯片的 MAC 地址作为硬件的唯一标志，此方法简单有效。然而，该方式存在一定缺陷，MAC 地址可被伪造，进而可能导致通信协议遭受攻击。此外，当用户赠与或退还设备后，只能由其本人从账号中删除设备，此举带来极大不便。在后续版本中，将采用软件生成的 UUID 作为设备唯一标志存储于 Flash 中。重新烧录完整镜像固件，可抹除该 UUID 信息，使设备变为全新状态。因此，用户若重新使用该设备，需在后台重新添加。

## 添加设备逻辑前置
当前，需按下 Boot 键或通过语言唤醒后，经由协议建立语音通道，方可播报添加设备的验证码。新版本将把验证码前置至 OTA 检查新固件接口，新设备只要完成网络连接，便应播报添加设备的语音与验证码。

## 讨论

欢迎提出建议。</div>

ipv6next2000, 2025-02-17:
这2条建议挺好。有个小问题，如果添加设备逻辑前置（只要完成网络连接），那么播报验证码，是只播报一次吗？如果想再次听，再按下boot键，是这个逻辑吧

78, 2025-02-18:
> 这2条建议挺好。有个小问题，如果添加设备逻辑前置（只要完成网络连接），那么播报验证码，是只播报一次吗？如果想再次听，再按下boot键，是这个逻辑吧

可以看代码，目前每分钟会重新检测一次。按boot会重启，连网后立即检测。

closed 2025-03-21 by 78

xiangxin1025, 2025-09-10:
1.8.6版本的固件：
现在我尝试使用 Flash Download Tool 把镜像固件写入到地址 0 ，实际并没有进行擦除UUID ，因为不需要重新绑定设备到服务器，这个是什么原因呢？