78/xiaozhi-esp32 #167: 4M Flash版本ESP32-S3模组顺利跑起，有报错

molun, 2025-02-15:
[image]

[image]
`I (1007889) Application: STATE: listening
I (1009879) Application: STATE: idle
I (1010039) Application: >> 今天天气怎么样？
I (1010089) Application: STATE: speaking
I (1010089) AudioCodec: Set output enable to true
I (1011389) Application: << 今天西宁天气多云，白天最高气温1度，夜间最低气温-10度。
W (1012299) MQTT: Received audio packet with wrong sequence: 34, expected: 32
W (1012779) MQTT: Received audio packet with wrong sequence: 42, expected: 36
W (1013259) MQTT: Received audio packet with wrong sequence: 50, expected: 44
W (1013799) MQTT: Received audio packet with wrong sequence: 59, expected: 53
W (1014039) MQTT: Received audio packet with wrong sequence: 63, expected: 62
W (1014279) MQTT: Received audio packet with wrong sequence: 67, expected: 64
W (1014459) MQTT: Received audio packet with wrong sequence: 75, expected: 74
I (1015209) Application: << 风力不大，南风2级。
I (1015289) main: Free internal: 10060 minimal internal: 416
I (1017789) Application: << 空气质量和湿度都还不错，适合出门走走哦！
I (1022459) Application: STATE: idle
I (1025289) main: Free internal: 22540 minimal internal: 416
I (1033519) AudioCodec: Set output enable to false
I (1035289) main: Free internal: 22540 minimal internal: 416
I (1045289) main: Free internal: 22540 minimal internal: 416
I (1055289) main: Free internal: 22540 minimal internal: 416
I (1065289) main: Free internal: 22540 minimal internal: 416
I (1075289) main: Free internal: 22540 minimal internal: 416
I (1083469) wifi:Set ps type: 1, coexist: 0

I (1085289) main: Free internal: 27948 minimal internal: 416
I (1095289) main: Free internal: 27948 minimal internal: 416
I (1105289) main: Free internal: 27948 minimal internal: 416
I (1115289) main: Free internal: 27948 minimal internal: 416
I (1125289) main: Free internal: 27948 minimal internal: 416
I (1135289) main: Free internal: 27948 minimal internal: 416
I (1145289) main: Free internal: 27908 minimal internal: 416
I (1155289) main: Free internal: 27948 minimal internal: 416
I (1165289) main: Free internal: 27948 minimal internal: 416
I (1175289) main: Free internal: 27908 minimal internal: 416
I (1185289) main: Free internal: 27948 minimal internal: 416
I (1195289) main: Free internal: 27948 minimal internal: 416
I (1205289) main: Free internal: 27948 minimal internal: 416
I (1215289) main: Free internal: 27948 minimal internal: 416
I (1225289) main: Free internal: 27948 minimal internal: 416
I (1235289) main: Free internal: 27948 minimal internal: 416
I (1236559) Application: STATE: connecting
I (1236729) MQTT: Session ID: 54d88427
I (1236739) wifi:Set ps type: 0, coexist: 0

I (1236739) Application: STATE: listening
I (1238249) Application: STATE: idle
I (1238419) Application: >> 唱首歌吧。
I (1238459) Application: STATE: speaking
I (1238469) AudioCodec: Set output enable to true
I (1238959) Application: << 好呀，你想听什么歌呢？
E (1240079) esp-aes: Failed to allocate memory
E (1240079) MQTT: Failed to decrypt audio data, ret: -1
W (1240079) MQTT: Received audio packet with wrong sequence: 32, expected: 31
E (1240089) esp-aes: Failed to allocate memory
E (1240089) MQTT: Failed to decrypt audio data, ret: -1
W (1240319) MQTT: Received audio packet with wrong sequence: 33, expected: 31
W (1240709) MQTT: Received audio packet with wrong sequence: 39, expected: 38
E (1240709) Dynamic Impl: alloc(4437 bytes) failed
E (1240709) esp-tls-mbedtls: write error :-0x7F00
E (1240709) transport_base: esp_tls_conn_write error, errno=Connection already in progress
E (1240719) mqtt_client: Writing failed: errno=119
I (1240719) esp_mqtt: MQTT error occurred: ESP_ERR_MBEDTLS_SSL_WRITE_FAILED
E (1240729) mqtt_client: Error write qos msg repsonse, qos = 1
W (1241159) MQTT: Received audio packet with wrong sequence: 42, expected: 41
W (1241459) MQTT: Received audio packet with wrong sequence: 49, expected: 46
W (1241709) MQTT: Received audio packet with wrong sequence: 51, expected: 50
W (1241879) MQTT: Received audio packet with wrong sequence: 59, expected: 58
I (1245289) main: Free internal: 26500 minimal internal: 416
I (1255289) main: Free internal: 26500 minimal internal: 416
I (1255929) esp-x509-crt-bundle: Certificate validated
I (1265289) main: Free internal: 22448 minimal internal: 416
I (1275289) main: Free internal: 22448 minimal internal: 416`

molun, 2025-02-15:
是不是还是内存小的锅

molun, 2025-02-15:
`I (11900) Application: >> 讲一个笑话。
I (11940) Application: STATE: speaking
I (12830) Application: << 好呀，等我想到好笑的就讲给你听。

abort() was called at PC 0x421000a3 on core 1
--- 0x421000a3: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48

Backtrace: 0x40379d19:0x3fcb5d90 0x40381dfd:0x3fcb5db0 0x403897f5:0x3fcb5dd0 0x421000a3:0x3fcb5e40 0x421000d8:0x3fcb5e60 0x421001b3:0x3fcb5e80 0x42100246:0x3fcb5ea0 0x4200d021:0x3fcb5ec0 0x4200d036:0x3fcb5ee0 0x4200d06d:0x3fcb5f00 0x4200d08d:0x3fcb5f20 0x4213f9b7:0x3fcb5f60 0x4201473d:0x3fcb5f80 0x42014775:0x3fcb5fc0 0x4200d632:0x3fcb5fe0 0x420168d9:0x3fcb6000 0x42016931:0x3fcb6030 0x4200d632:0x3fcb6050 0x42016ae2:0x3fcb6070 0x42016b31:0x3fcb60b0 0x403828f1:0x3fcb60d0
--- 0x40379d19: panic_abort at D:/MOLUN/Espressif/frameworks/esp-idf-v5.4/components/esp_system/panic.c:454
0x40381dfd: esp_system_abort at D:/MOLUN/Espressif/frameworks/esp-idf-v5.4/components/esp_system/port/esp_system_chip.c:92
0x403897f5: abort at D:/MOLUN/Espressif/frameworks/esp-idf-v5.4/components/newlib/abort.c:38
0x421000a3: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48     
0x421000d8: std::terminate() at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:58 (discriminator 1)
0x421001b3: __cxa_throw at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:98
0x42100246: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:54 (discriminator 2)  
0x4200d021: std::__new_allocator<long>::allocate(unsigned int, void const*) at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/new_allocator.h:151
0x4200d036: std::allocator<long>::allocate(unsigned int) at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/allocator.h:196
 (inlined by) std::allocator_traits<std::allocator<long> >::allocate(std::allocator<long>&, unsigned int) at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/alloc_traits.h:478
 (inlined by) std::_Vector_base<long, std::allocator<long> >::_M_allocate(unsigned int) at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:380
 (inlined by) std::_Vector_base<long, std::allocator<long> >::_M_create_storage(unsigned int) at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:398
0x4200d06d: std::_Vector_base<long, std::allocator<long> >::_Vector_base(unsigned int, std::allocator<long> const&) at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:334
 (inlined by) std::vector<long, std::allocator<long> >::vector(unsigned int, std::allocator<long> const&) at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:557
0x4200d08d: NoAudioCodec::Write(short const*, int) at D:/MOLUN/Documents/GitHub/xiaozhi-esp32/main/audio_codecs/no_audio_codec.cc:338 (discriminator 1) 
0x4213f9b7: AudioCodec::OutputData(std::vector<short, std::allocator<short> >&) at D:/MOLUN/Documents/GitHub/xiaozhi-esp32/main/audio_codecs/audio_codec.cc:26 (discriminator 2)
0x4201473d: Application::OutputAudio()::{lambda()#1}::operator()() at D:/MOLUN/Documents/GitHub/xiaozhi-esp32/main/application.cc:531
0x42014775: std::_Function_handler<void (), Application::OutputAudio()::{lambda()#1}>::_M_invoke(std::_Any_data const&) at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/invoke.h:61
 (inlined by) __invoke_r<void, Application::OutputAudio()::<lambda()>&> at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/invoke.h:111
 (inlined by) _M_invoke at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/std_function.h:290
0x4200d632: std::function<void ()>::operator()() const at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/std_function.h:591
0x420168d9: BackgroundTask::Schedule(std::function<void ()>)::{lambda()#1}::operator()() const at D:/MOLUN/Documents/GitHub/xiaozhi-esp32/main/background_task.cc:27
0x42016931: std::_Function_handler<void (), BackgroundTask::Schedule(std::function<void ()>)::{lambda()#1}>::_M_invoke(std::_Any_data const&) at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/invoke.h:61
 (inlined by) __invoke_r<void, BackgroundTask::Schedule(std::function<void()>)::<lambda()>&> at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/invoke.h:111
 (inlined by) _M_invoke at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/std_function.h:290
[... cut ...]
I (170) Board: UUID: 66b7ae41-4e5a-4caf-9f1b-64b81421ea40
I (170) button: IoT Button Version: 3.5.0
I (170) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (180) button: IoT Button Version: 3.5.0
I (180) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (190) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (200) Application: STATE: starting
I (200) NoAudioCodec: Simplex channels created`

78, 2025-02-15:
没有PSRAM吗？你是开发板吗？模型选的是哪个

molun, 2025-02-16:
> 没有PSRAM吗？你是开发板吗？模型选的是哪个

没有PSRAM，自己模组制作的，选取面包板WIFI这个板型，编译关闭PSRAM、语音唤醒、选取4M固件和分区表编译。

78, 2025-02-16:
试试用Websocket协议。

molun, 2025-02-17:
> 试试用Websocket协议。

[image]
这里的token在哪里获取 还是保持这样就可以

molun, 2025-02-17:
```
ELF file SHA256: 590f6848d

Rebooting...
���I (36) cpu_start: Multicore app
I (45) cpu_start: Pro cpu start user code
I (45) cpu_start: cpu freq: 240000000 Hz
I (46) app_init: Application information:
I (46) app_init: Project name:     xiaozhi
I (49) app_init: App version:      1.1.9
I (53) app_init: Compile time:     Feb 17 2025 15:32:10
I (58) app_init: ELF file SHA256:  590f6848d...
I (62) app_init: ESP-IDF:          v5.4-dirty
I (66) efuse_init: Min chip rev:     v0.0
I (70) efuse_init: Max chip rev:     v0.99
I (74) efuse_init: Chip rev:         v0.2
I (78) heap_init: Initializing. RAM available for dynamic allocation:
I (84) heap_init: At 3FCA6BB0 len 00042B60 (266 KiB): RAM
I (89) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (94) heap_init: At 600FE11C len 00001EBC (7 KiB): RTCRAM
I (100) spi_flash: detected chip: gd
I (102) spi_flash: flash io: qio
I (106) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (112) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (118) main_task: Started on CPU0
I (158) main_task: Calling app_main()
I (178) BackgroundTask: background_task started
I (178) Board: UUID: 9ce89641-df19-42bc-a543-2f3059b783dc
I (178) button: IoT Button Version: 3.5.0
I (178) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (188) button: IoT Button Version: 3.5.0
I (188) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (198) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (208) Application: STATE: starting
I (218) NoAudioCodec: Simplex channels created
I (218) Ssd1306Display: Initialize LVGL
I (218) LVGL: Starting LVGL task
I (218) Ssd1306Display: Install SSD1306 driver
I (228) Ssd1306Display: SSD1306 driver installed
I (228) Ssd1306Display: Turning display on
I (338) Ssd1306Display: Adding LCD screen
I (368) Application: WiFi board detected, setting opus encoder complexity to 3
I (368) AudioCodec: Set input enable to true
I (368) AudioCodec: Set output enable to true
I (388) pp: pp rom version: e7ae62f
I (388) net80211: net80211 rom version: e7ae62f
I (398) wifi:wifi driver task: 3fcd4d18, prio:23, stack:6656, core=0
I (398) wifi:wifi firmware version: 48ea317a7
I (398) wifi:wifi certification version: v7.0
I (398) wifi:config NVS flash: disabled
I (408) wifi:config nano formatting: disabled
I (408) wifi:Init data frame dynamic rx buffer num: 32
I (408) wifi:Init static rx mgmt buffer num: 5
I (418) wifi:Init management short buffer num: 32
I (418) wifi:Init static tx buffer num: 16
I (428) wifi:Init static tx FG buffer num: 2
I (428) wifi:Init static rx buffer size: 1600
I (428) wifi:Init static rx buffer num: 16
I (438) wifi:Init dynamic rx buffer num: 32
I (438) wifi_init: rx ba win: 16
I (448) wifi_init: accept mbox: 6
I (448) wifi_init: tcpip mbox: 32
I (448) wifi_init: udp mbox: 6
I (448) wifi_init: tcp mbox: 6
I (458) wifi_init: tcp tx win: 5760
I (458) wifi_init: tcp rx win: 5760
I (458) wifi_init: tcp mss: 1440
I (468) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
I (498) wifi:mode : sta (34:b7:da:6b:a1:9c)
I (508) wifi:enable tsf
[... 29 lines cut ...]

Backtrace: 0x40379d19:0x3fcecae0 0x40381dfd:0x3fcecb00 0x403897f5:0x3fcecb20 0x42100813:0x3fcecb90 0x42100848:0x3fcecbb0 0x42100923:0x3fcecbd0 0x421009b6:0x3fcecbf0 0x4200c9a5:0x3fcecc10 0x4200cae6:0x3fcecc30 0x42011700:0x3fcecc60 0x42013be9:0x3fcecc80 0x42013e35:0x3fcecd90 0x42013f31:0x3fceceb0 0x403828f1:0x3fceced0
--- 0x40379d19: panic_abort at D:/MOLUN/Espressif/frameworks/esp-idf-v5.4/components/esp_system/panic.c:454
0x40381dfd: esp_system_abort at D:/MOLUN/Espressif/frameworks/esp-idf-v5.4/components/esp_system/port/esp_system_chip.c:92
0x403897f5: abort at D:/MOLUN/Espressif/frameworks/esp-idf-v5.4/components/newlib/abort.c:38
0x42100813: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48     
0x42100848: std::terminate() at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:58 (discriminator 1)      
0x42100923: __cxa_throw at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:98
0x421009b6: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:54 (discriminator 2)  
0x4200c9a5: std::__new_allocator<char>::allocate(unsigned int, void const*) at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/new_allocator.h:151
 (inlined by) std::allocator<char>::allocate(unsigned int) at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/allocator.h:196
 (inlined by) std::allocator_traits<std::allocator<char> >::allocate(std::allocator<char>&, unsigned int) at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/alloc_traits.h:478
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_S_allocate(std::allocator<char>&, unsigned int) at D:/MOLUN/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/basic_string.h:131
[... cut ...]
0x42013e35: Application::CheckNewVersion() at D:/MOLUN/Documents/GitHub/xiaozhi-esp32/main/application.cc:106
0x42013f31: Application::Start()::{lambda(void*)#2}::_FUN(void*) at D:/MOLUN/Documents/GitHub/xiaozhi-esp32/main/application.cc:407
 (inlined by) _FUN at D:/MOLUN/Documents/GitHub/xiaozhi-esp32/main/application.cc:409
0x403828f1: vPortTaskWrapper at D:/MOLUN/Espressif/frameworks/esp-idf-v5.4/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

```

1.1.9 会不断重启 MQTT和Websocket都试过，Websocket会稍微好一点 烧录完毕后台删除设备，重启准备重新添加就一直重启或者显示待命 唤醒就重启

78, 2025-02-17:
@ooxxU

molun, 2025-02-17:
> 你有改动代码吗？怎么可用的SRAM这么小。

没有，只调整了几个不同的GPIO，其余部分未改动，退回1.1.2的代码情况会好些

78, 2025-02-18:
拉一下最新的试试，启用了wifi动态内存，节约了5kb的内存。

molun, 2025-02-18:
> 拉一下最新的试试，启用了wifi动态内存，节约了5kb的内存。

使用凌晨1点多的代码编译，和@ooxxU沟通过，不接屏幕使用正常，对话流畅。插入0.96屏幕后对话很大概率会导致重启和卡死。
以下日志是不插屏幕
`I (28) cpu_start: Multicore app
I (37) cpu_start: Pro cpu start user code
I (37) cpu_start: cpu freq: 240000000 Hz
I (37) app_init: Application information:
I (37) app_init: Project name:     xiaozhi
I (40) app_init: App version:      1.1.9
I (44) app_init: Compile time:     Feb 18 2025 01:23:13
I (49) app_init: ELF file SHA256:  4629b2121...
I (53) app_init: ESP-IDF:          v5.4-dirty
I (57) efuse_init: Min chip rev:     v0.0
I (61) efuse_init: Max chip rev:     v0.99 
I (65) efuse_init: Chip rev:         v0.2
I (69) heap_init: Initializing. RAM available for dynamic allocation:
I (75) heap_init: At 3FCA6BB0 len 00042B60 (266 KiB): RAM
I (80) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (85) heap_init: At 600FE11C len 00001EBC (7 KiB): RTCRAM
I (91) spi_flash: detected chip: gd
I (93) spi_flash: flash io: qio
I (97) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (103) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (109) main_task: Started on CPU0
I (149) main_task: Calling app_main()
I (169) BackgroundTask: background_task started
I (169) Board: UUID: 6c0ae5e9-74db-4505-8d26-054670e79957
I (169) button: IoT Button Version: 3.5.0
I (169) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (179) button: IoT Button Version: 3.5.0
I (179) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (189) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

I (199) Application: STATE: starting
I (199) NoAudioCodec: Simplex channels created
I (209) Ssd1306Display: Initialize LVGL
I (209) LVGL: Starting LVGL task
I (209) Ssd1306Display: Install SSD1306 driver
I (219) Ssd1306Display: SSD1306 driver installed
E (219) i2c.master: I2C hardware NACK detected
E (229) i2c.master: I2C transaction unexpected nack detected
E (229) i2c.master: s_i2c_synchronous_transaction(918): I2C transaction failed
E (239) i2c.master: i2c_master_multi_buffer_transmit(1180): I2C transaction failed
E (249) lcd_panel.io.i2c: panel_io_i2c_tx_buffer(193): i2c transaction failed
E (249) lcd_panel.ssd1306: panel_ssd1306_init(151): io tx param SSD1306_CMD_SET_MULTIPLEX failed
E (259) Ssd1306Display: Failed to initialize display
I (269) Application: WiFi board detected, setting opus encoder complexity to 3
I (269) AudioCodec: Set input enable to true
I (279) AudioCodec: Set output enable to true
I (279) pp: pp rom version: e7ae62f
I (279) net80211: net80211 rom version: e7ae62f
I (299) wifi:wifi driver task: 3fccd584, prio:23, stack:6656, core=0
I (299) wifi:wifi firmware version: 48ea317a7
I (299) wifi:wifi certification version: v7.0
I (299) wifi:config NVS flash: disabled
I (309) wifi:config nano formatting: disabled
I (309) wifi:Init data frame dynamic rx buffer num: 32
I (309) wifi:Init static rx mgmt buffer num: 5
I (319) wifi:Init management short buffer num: 32
I (319) wifi:Init static tx buffer num: 16
I (329) wifi:Init static tx FG buffer num: 2
I (329) wifi:Init static rx buffer size: 1600
I (339) wifi:Init static rx buffer num: 16
I (339) wifi:Init dynamic rx buffer num: 32
I (339) wifi_init: rx ba win: 16
I (349) wifi_init: accept mbox: 6
I (349) wifi_init: tcpip mbox: 32
I (349) wifi_init: udp mbox: 6
I (349) wifi_init: tcp mbox: 6
[... 963 lines cut ...]
W (87770) MQTT: Received audio packet with wrong sequence: 52, expected: 51

[2025-02-18 12:17:42.202]# RECV ASCII>
W (88070) MQTT: Received audio packet with wrong sequence: 59, expected: 55

[2025-02-18 12:17:42.683]# RECV ASCII>
W (88550) MQTT: Received audio packet with wrong sequence: 65, expected: 62

[2025-02-18 12:17:43.151]# RECV ASCII>
W (89030) MQTT: Received audio packet with wrong sequence: 73, expected: 70

[2025-02-18 12:17:49.314]# RECV ASCII>
I (95190) main: Free internal: 25528 minimal internal: 200

[2025-02-18 12:17:56.010]# RECV ASCII>
I (101880) esp-x509-crt-bundle: Certificate validated

[2025-02-18 12:17:59.312]# RECV ASCII>
I (105190) main: Free internal: 21504 minimal internal: 200

`

[image]

https://github.com/user-attachments/assets/7c3ecdd0-43d2-4608-b586-debb8b2d7244

molun, 2025-02-18:
`I (244370) main: Free internal: 18580 minimal internal: 468
I (245070) Application: STATE: listening
I (252420) Application: >> 如果我要复刻一个虾个小智AI机器人的话，应该注意哪些方面？
I (252480) Application: STATE: speaking
I (252900) Application: << 复刻一个AI机器人像虾哥小智的话，有几点需要注意：

I (254370) main: Free internal: 16016 minimal internal: 468
I (257530) Application: << 一， 明确目标 ：了解你想要实现的核心功能是什么。
I (263400) Application: << 
二， 收集资料 ：研究虾哥小智的特性，收集相关技术文档。
I (264370) main: Free internal: 16752 minimal internal: 468
I (270720) Application: << 
三， 技术选型 ：选择合适的开发平台和语言，比如Python和Flask框架。
I (274370) main: Free internal: 15460 minimal internal: 468
I (278290) Application: << 
四， 功能设计 ：设计机器人的交互逻辑，包括对话管理和任务执行。
I (284370) main: Free internal: 14516 minimal internal: 468
I (286450) Application: << 
五， 训练模型 ：准备数据集，训练自然语言处理模型。
I (292800) Application: << 
六， 测试优化 ：不断测试并优化机器人的表现。
I (294370) main: Free internal: 17624 minimal internal: 468
I (298690) Application: << 
七， 用户反馈 ：上线后收集用户反馈，持续改进。
I (304370) main: Free internal: 17352 minimal internal: 468
I (304440) Application: << 

你对哪个方面最感兴趣？
I (306780) Application: << 我们可以详细讨论一下。
I (309870) Application: STATE: listening
I (314370) main: Free internal: 28272 minimal internal: 468
I (314580) Application: >> 哎呀，说不得了，应该找出来是。
I (314640) Application: STATE: speaking
I (315510) Application: << 没问题，我们可以从最基础的开始。
I (318370) Application: << 比如，你可以先了解一下虾哥小智的基本功能和特点，
E (321430) esp-aes: Failed to allocate memory
E (321430) esp-tls-mbedtls: read error :-0x0001
E (321430) TlsTransport: TLS读取失败: -1
I (321430) WS: Websocket disconnected
I (321430) wifi:Set ps type: 1, coexist: 0

I (321470) Application: STATE: idle
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x00060f23  PS      : 0x00060d30  A0      : 0x82015f65  A1      : 0x3fcb60f0
A2      : 0x3fce5d14  A3      : 0x3fcae1d8  A4      : 0x3c16774c  A5      : 0x000000aa  
A6      : 0x3fcb6110  A7      : 0x3fcb6110  A8      : 0x8200d771  A9      : 0x3fcb60d0
A10     : 0x3fce5d14  A11     : 0x00000018  A12     : 0x80382ca9  A13     : 0x3fcac470  
A14     : 0x00060023  A15     : 0x00000003  SAR     : 0x00000019  EXCCAUSE: 0x00000014
EXCVADDR: 0x00060f20  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  
--- 0x40056f5c: memcpy in ROM
0x40056f72: memcpy in ROM

Backtrace: 0x00060f20:0x3fcb60f0 0x42015f62:0x3fcb6110 0x42015fb1:0x3fcb6150 0x403828f1:0x3fcb6170
--- 0x42015f62: BackgroundTask::BackgroundTaskLoop() at D:/MOLUN/Documents/GitHub/xiaozhi-esp32/main/background_task.cc:59
0x42015fb1: BackgroundTask::BackgroundTask(unsigned long)::{lambda(void*)#1}::_FUN(void*) at D:/MOLUN/Documents/GitHub/xiaozhi-esp32/main/background_task.cc:10
 (inlined by) _FUN at D:/MOLUN/Documents/GitHub/xiaozhi-esp32/main/background_task.cc:11
0x403828f1: vPortTaskWrapper at D:/MOLUN/Espressif/frameworks/esp-idf-v5.4/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: 4f78bdacb

Rebooting...
���I (29) cpu_start: Multicore app
I (37) cpu_start: Pro cpu start user code
I (37) cpu_start: cpu freq: 240000000 Hz
I (38) app_init: Application information:
I (38) app_init: Project name:     xiaozhi
I (41) app_init: App version:      1.2.1
I (45) app_init: Compile time:     Feb 18 2025 17:45:15
[... 40 lines cut ...]
I (400) wifi:Init data frame dynamic rx buffer num: 32
I (400) wifi:Init static rx mgmt buffer num: 5
I (410) wifi:Init management short buffer num: 32
I (410) wifi:Init static tx buffer num: 16
I (410) wifi:Init static tx FG buffer num: 2
I (420) wifi:Init static rx buffer size: 1600
I (420) wifi:Init static rx buffer num: 16
I (430) wifi:Init dynamic rx buffer num: 32
I (430) wifi_init: rx ba win: 16
I (430) wifi_init: accept mbox: 6
I (440) wifi_init: tcpip mbox: 32
I (440) wifi_init: udp mbox: 6
I (440) wifi_init: tcp mbox: 6
I (440) wifi_init: tcp tx win: 5760
I (450) wifi_init: tcp rx win: 5760
I (450) wifi_init: tcp mss: 1440
I (450) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
I (490) wifi:mode : sta (34:b7:da:6b:a1:9c)
I (490) wifi:enable tsf
I (2900) wifi: Found AP: ChinaNet-品味酒行, BSSID: 7c:da:c3:b3:c1:c5, RSSI: -55, Channel: 2, Authmode: 4
W (2900) wifi:Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
I (3240) wifi:new:<2,0>, old:<1,0>, ap:<255,255>, sta:<2,0>, prof:1, snd_ch_cfg:0x0
I (3240) wifi:state: init -> auth (0xb0)
I (3270) wifi:state: auth -> assoc (0x0)
I (3280) wifi:state: assoc -> run (0x10)`

78, 2025-02-18:
不接屏幕的问题，今天1.2.1新发布的版本已经修复了。如果使用中还有其他问题，可能需要 @ooxxU 来看

closed 2025-03-21 by 78