78/xiaozhi-esp32 #168: 有人在MacOS上编译成功的吗

chemdemo, 2025-02-15:
报错信息：
```shell
/Users/dm/esp/xiaozhi-esp32/main/audio_codecs/tcircles3_audio_codec.cc: In member function 'void Tcircles3AudioCodec::CreateVoiceHardware(gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t)':
/Users/dm/esp/xiaozhi-esp32/main/audio_codecs/tcircles3_audio_codec.cc:71:5: error: designator order for field 'i2s_std_clk_config_t::ext_clk_freq_hz' does not match declaration order in 'i2s_std_clk_config_t'
   71 |     };
      |     ^
/Users/dm/esp/xiaozhi-esp32/main/audio_codecs/tcircles3_audio_codec.cc:88:5: error: designator order for field 'i2s_std_clk_config_t::ext_clk_freq_hz' does not match declaration order in 'i2s_std_clk_config_t'
   88 |     };
      |     ^
[1838/1867] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/iot/thing.cc.obj
ninja: build stopped: subcommand failed.
```

dhlsam, 2025-02-15:
sdk5.5编译会出错，需要5.3可以成功

78, 2025-02-16:
应该是sdk5.5有兼容性问题

yusuhua, 2025-02-23:
> 目标：esp32s3 开发板
> 
> 报错信息：
> 
> [1825/1867] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/audio_codecs/tcircles3_audio_codec.cc.obj
> FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/audio_codecs/tcircles3_audio_codec.cc.obj 
> /Users/dm/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ -DBOARD_TYPE=\"bread-compact-wifi\" -DESP_PLATFORM -DIDF_VER=\"v5.5-dev-1860-g0d6099ec53\" -DLV_CONF_INCLUDE_SIMPLE -DLV_LVGL_H_INCLUDE_SIMPLE -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -DUNITY_INCLUDE_CONFIG_H -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/dm/esp/xiaozhi-esp32/build/config -I/Users/dm/esp/xiaozhi-esp32/main -I/Users/dm/esp/xiaozhi-esp32/main/display -I/Users/dm/esp/xiaozhi-esp32/main/audio_codecs -I/Users/dm/esp/xiaozhi-esp32/main/protocols -I/Users/dm/esp/xiaozhi-esp32/main/audio_processing -I/Users/dm/esp/xiaozhi-esp32/main/boards/common -I/Users/dm/esp/esp-idf/components/newlib/platform_include -I/Users/dm/esp/esp-idf/components/freertos/config/include -I/Users/dm/esp/esp-idf/components/freertos/config/include/freertos -I/Users/dm/esp/esp-idf/components/freertos/config/xtensa/include -I/Users/dm/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/dm/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/dm/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/Users/dm/esp/esp-idf/components/freertos/esp_additions/include -I/Users/dm/esp/esp-idf/components/esp_hw_support/include -I/Users/dm/esp/esp-idf/components/esp_hw_support/include/soc -I/Users/dm/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/Users/dm/esp/esp-idf/components/esp_hw_support/dma/include -I/Users/dm/esp/esp-idf/components/esp_hw_support/ldo/include -I/Users/dm/esp/esp-idf/components/esp_hw_support/debug_probe/include -I/Users/dm/esp/esp-idf/components/esp_hw_support/mspi_timing_tuning/include -I/Users/dm/esp/esp-idf/components/esp_hw_support/power_supply/include -I/Users/dm/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -I/Users/dm/esp/esp-idf/components/esp_hw_support/port/esp32s3/include -I/Users/dm/esp/esp-idf/components/esp_hw_support/mspi_timing_tuning/port/esp32s3/. -I/Users/dm/esp/esp-idf/components/esp_hw_support/mspi_timing_tuning/port/esp32s3/include -I/Users/dm/esp/esp-idf/components/heap/include -I/Users/dm/esp/esp-idf/components/heap/tlsf -I/Users/dm/esp/esp-idf/components/log/include -I/Users/dm/esp/esp-idf/components/soc/include -I/Users/dm/esp/esp-idf/components/soc/esp32s3 -I/Users/dm/esp/esp-idf/components/soc/esp32s3/include -I/Users/dm/esp/esp-idf/components/soc/esp32s3/register -I/Users/dm/esp/esp-idf/components/hal/platform_port/include -I/Users/dm/esp/esp-idf/components/hal/esp32s3/include -I/Users/dm/esp/esp-idf/components/hal/include -I/Users/dm/esp/esp-idf/components/esp_rom/include -I/Users/dm/esp/esp-idf/components/esp_rom/esp32s3/include -I/Users/dm/esp/esp-idf/components/esp_rom/esp32s3/include/esp32s3 -I/Users/dm/esp/esp-idf/components/esp_rom/esp32s3 -I/Users/dm/esp/esp-idf/components/esp_common/include -I/Users/dm/esp/esp-idf/components/esp_system/include -I/Users/dm/esp/esp-idf/components/esp_system/port/soc -I/Users/dm/esp/esp-idf/components/esp_system/port/include/private -I/Users/dm/esp/esp-idf/components/xtensa/esp32s3/include -I/Users/dm/esp/esp-idf/components/xtensa/include -I/Users/dm/esp/esp-idf/components/xtensa/deprecated_include -I/Users/dm/esp/esp-idf/components/esp_timer/include -I/Users/dm/esp/esp-idf/components/lwip/include -I/Users/dm/esp/esp-idf/components/lwip/include/apps -I/Users/dm/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/dm/esp/esp-idf/components/lwip/lwip/src/include -I/Users/dm/esp/esp-idf/components/lwip/port/include -I/Users/dm/esp/esp-idf/components/lwip/port/freertos/include -I/Users/dm/esp/esp-idf/components/lwip/port/esp32xx/include -I/Users/dm/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/Users/dm/esp/esp-idf/components/lwip/port/esp32xx/include/sys -I/Users/dm/esp/xiaozhi-esp32/managed_components/78__esp-ml307/include -I/Users/dm/esp/esp-idf/components/esp_driver_gpio/include -I/Users/dm/esp/esp-idf/components/esp_driver_uart/include -I/Users/dm/esp/esp-idf/components/vfs/include -I/Users/dm/esp/esp-idf/components/esp-tls -I/Users/dm/esp/esp-idf/components/esp-tls/esp-tls-crypto -I/Users/dm/esp/esp-idf/components/mbedtls/port/include -I/Users/dm/esp/esp-idf/components/mbedtls/mbedtls/include -I/Users/dm/esp/esp-idf/components/mbedtls/mbedtls/library -I/Users/dm/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/Users/dm/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -I/Users/dm/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -I/Users/dm/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -I/Users/dm/esp/esp-idf/components/esp_http_client/include -I/Users/dm/esp/esp-idf/components/esp_event/include -I/Users/dm/esp/esp-idf/components/mqtt/esp-mqtt/include -I/Users/dm/esp/esp-idf/components/tcp_transport/include -I/Users/dm/esp/xiaozhi-esp32/managed_components/78__esp-opus-encoder/include -I/Users/dm/esp/xiaozhi-esp32/managed_components/78__esp-wifi-connect/include -I/Users/dm/esp/esp-idf/components/esp_http_server/include -I/Users/dm/esp/esp-idf/components/http_parser -I/Users/dm/esp/esp-idf/components/esp_wifi/ [... cut ...]
> /Users/dm/esp/xiaozhi-esp32/main/audio_codecs/tcircles3_audio_codec.cc:71:5: error: designator order for field 'i2s_std_clk_config_t::ext_clk_freq_hz' does not match declaration order in 'i2s_std_clk_config_t'
>    71 |     };
>       |     ^
> /Users/dm/esp/xiaozhi-esp32/main/audio_codecs/tcircles3_audio_codec.cc:88:5: error: designator order for field 'i2s_std_clk_config_t::ext_clk_freq_hz' does not match declaration order in 'i2s_std_clk_config_t'
>    88 |     };
>       |     ^
> [1838/1867] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/iot/thing.cc.obj
> ninja: build stopped: subcommand failed.

试下这个呢？https://github.com/78/xiaozhi-esp32/issues/173

yusuhua, 2025-02-23:
试下这个呢？https://github.com/78/xiaozhi-esp32/issues/173

bingcheng1998, 2025-03-15:
可以先试试直接在macos上烧录编译好的固件。
烧录教程：https://github.com/espressif/esptool/issues/908#issuecomment-2726138451

closed 2025-03-21 by 78