78/xiaozhi-esp32 #170: 服务断的AI agent是使用什么框架开发的？

leleZeng, 2025-02-16:
你好作者，我想了解下您服务端的AI agent采用什么框架开发的，是否使用了类似Dify的开源系统？此外您使用的TTS模型看起来效果非常好，我去火山TTS模型并没有找到这个音色的模型，是否使用了其他的TTS模型呢？

Huang-junsen, 2025-02-17:
火山（豆包）的大模型语音合成->湾湾小何的语音，相关实现应该是本地部署的文本模型和付费的豆包语音合成大模型，实时返回后给tts合成再返回，不过我好奇为什么响应那么快

leleZeng, 2025-02-18:
> 火山（豆包）的大模型语音合成->湾湾小何的语音，相关实现应该是本地部署的文本模型和付费的豆包语音合成大模型，实时返回后给tts合成再返回，不过我好奇为什么响应那么快

@Huang-junsen 我在豆包上没有找到湾湾小何的音色啊，只有甜美台妹的音色。

Huang-junsen, 2025-02-18:
> > 火山（豆包）的大模型语音合成->湾湾小何的语音，相关实现应该是本地部署的文本模型和付费的豆包语音合成大模型，实时返回后给tts合成再返回，不过我好奇为什么响应那么快
> 
> [@Huang-junsen](https://github.com/Huang-junsen) 我在豆包上没有找到湾湾小何的音色啊，只有甜美台妹的音色。

搜：豆包语音合成或者打开下面的链接往下滑就能看到了

https://www.volcengine.com/product/tts

[image]

leleZeng, 2025-02-18:
> > > 火山（豆包）的大模型语音合成->湾湾小何的语音，相关实现应该是本地部署的文本模型和付费的豆包语音合成大模型，实时返回后给tts合成再返回，不过我好奇为什么响应那么快
> > 
> > 
> > [@Huang-junsen](https://github.com/Huang-junsen) 我在豆包上没有找到湾湾小何的音色啊，只有甜美台妹的音色。
> 
> 搜：豆包语音合成或者打开下面的链接往下滑就能看到了
> 
> https://www.volcengine.com/product/tts
> 
> [image]

感谢，火山有大模型语音合成和语音合成两个产品，我找错产品了。

  小智AI响应快应该有多方面的原因，首先，语音识别和语音合成使用了双向流式调用的API，这样系统能够在用户说话时就开始将语音转为文字，而不是等到完全说完。这样可以大大提升识别速度。其次，LLM（大型语言模型）的输出速度快，因为它调用了外部API，语音合成也支持碎片化合成，这意味着生成的音频流也是分段传输的，系统只需要将这些音频片段转发给客户端，就能实现非常快的响应和语音播放。这样整体的即时语音效果非常好。  
  其实我有点不太明白，之前我用qwen模型的时候，它并不支持function call的流式输出。但小智AI却能支持function call，我测试过它能调用天气的函数和播放音乐。难道是小智AI自己部署的qwen大模型做了扩展，才有了这些功能吗？

Huang-junsen, 2025-02-18:
我看了后台里面历史记录、当我说今天天气怎么样？就会调一个get_weather的方法去获取，大概的流程是小智本地部署LLM，在prompt上做了调整、当用户说出关键字时会返回一个标识、当命中标识时就会去执行对应的方法、目前开源的第三方产品大部分都是这样处理的，还有一点，小智的除了双工通信外，它还有个mqtt的通信、回合制，问题是我们调小智接口时，我这一说完立马就能识别出文本然后发送给小智服务器、然后tts合成返回

可以看看我仓库里的[py-xiaozhi](https://github.com/Huang-junsen/py-xiaozhi.git)，脱离硬件的、在电脑上就能体验小智了

leleZeng, 2025-02-18:
哈哈 我就是看你了的py-xiaozhi里的通讯协议才猜测出他的设计流程，我正在开发一套java版本的server。

Huang-junsen, 2025-02-18:
> 哈哈 我就是看你了的py-xiaozhi里的通讯协议才猜测出他的设计流程，我正在开发一套java版本的server。

来都来了，点个start再走（滑稽），目前只对音频发送这块好奇，说完能立马识别出文本和生成录音、识别这块还是在本地

miniygq, 2025-03-06:
server端有开源吗

Huang-junsen, 2025-03-06:
目前官方小智后端没开源，有个第三方的esp32-server，目前1.4k start

closed 2025-03-21 by 78

qun0297, 2025-04-12:
这个就是火山的tts语音引擎，用的singalR之类的实时通讯，流式响应，没啥技术含量的，就是这个tts收费，不然自己弄了额，当然也有开源的tts库，但有些麻烦的。等过段时间有好的又免费的tts，自己做个软件就好了，没必要。