78/xiaozhi-esp32 #173: 项目构建报错FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/audio_codecs/tcircles3_audio_codec.cc.obj

luckyxiangfeng, 2025-02-17:
xiaozhi-esp32/main/audio_codecs/tcircles3_audio_codec.cc:88:5: error: designator order for field 'i2s_std_clk_config_t::ext_clk_freq_hz' does not match declaration order in 'i2s_std_clk_config_t'

78, 2025-02-17:
esp sdk版本号多少？

luckyxiangfeng, 2025-02-18:
> esp sdk版本号多少？

5.5

dengjb, 2025-02-19:
mac 编译通过后，运行报错：
E (362) i2c.master: I2C transaction unexpected nack detected
E (362) i2c.master: s_i2c_synchronous_transaction(888): I2C transaction failed
E (372) i2c.master: i2c_master_transmit(1116): I2C transaction failed
E (382) lcd_panel.io.i2c: panel_io_i2c_tx_buffer(179): i2c transaction failed
E (392) lcd_panel.ssd1306: panel_ssd1306_init(147): io tx param SSD1306_CMD_SET_MULTIPLEX failed
E (402) Ssd1306Display: Failed to initialize display
导致不能唤醒，请问有方法解决吗

dengjb, 2025-02-19:
> xiaozhi-esp32/main/audio_codecs/tcircles3_audio_codec.cc:88:5: error: designator order for field 'i2s_std_clk_config_t::ext_clk_freq_hz' does not match declaration order in 'i2s_std_clk_config_t'

需要切换到5.3

dengjb, 2025-02-19:
> mac 编译通过后，运行报错： E (362) i2c.master: I2C transaction unexpected nack detected E (362) i2c.master: s_i2c_synchronous_transaction(888): I2C transaction failed E (372) i2c.master: [...]

@78

yusuhua, 2025-02-21:
如果修改tcircles3_audio_codec.cc，希望可以将45引脚相关操作实现可配置化，有的设备驱动与t-circle-s3相同，却没有MAX98357A_SD_MODE类似引脚，复用起来比较麻烦。

78, 2025-02-21:
可以帮忙修改一下提交一个PR。

78, 2025-02-24:
建议测试后，提交PR，我合并进来。我手上没有这个设备。

closed 2025-05-20 by 78