78/xiaozhi-esp32 #196: 使用ESP32-S3面包板来控制开源闪电狗-如何添加IOT控制指令

fengjiannan2010, 2025-02-20:
实现原理：使用WIFI 面包版本，添加舵机驱动，添加语音指令控制
问题：
如何添加IOT指令，用来控制机器狗
灯光控制：开灯，关灯，呼吸灯，闪光灯，彩虹灯，流水灯，随机灯光等。
动作控制：立正，趴下，跳舞，左转，右转，摇摆，扭一扭等。
能否给予建议或指导，我查看了ESP官方文档，是否可以通过集成ESP-SR来实现语音唤醒，语音指令识别。

fengjiannan2010, 2025-02-28:
### 这个需要服务端支持吧，我现在本地语音识别，能识别到命令词了，就是逻辑需要调整

#### wake_word_detect.cc
```
#include "wake_word_detect.h"
#include "application.h"

#include <model_path.h>
#include <arpa/inet.h>
#include <sstream>

#define DETECTION_RUNNING_EVENT 1

static const char* TAG = "WakeWordDetect";

const char *cmd_phoneme[] = {
    "da kai kong tiao",   // command_id = 0
    "guan bi kong tiao",   // command_id = 1
    "da kai dian deng", // command_id = 2
    "guan bi dian deng" // command_id = 3
};

WakeWordDetect::WakeWordDetect()
    : afe_detection_data_(nullptr),
      multinet_model_data_(nullptr),
      wake_word_pcm_(),
      wake_word_opus_() {

    event_group_ = xEventGroupCreate();
}

WakeWordDetect::~WakeWordDetect() {
    if (afe_detection_data_ != nullptr) {
        afe_handle_->destroy(afe_detection_data_);
    }

    if(multinet_model_data_!=nullptr){
        multinet_iface_->destroy(multinet_model_data_);
    }

    if (wake_word_encode_task_stack_ != nullptr) {
        heap_caps_free(wake_word_encode_task_stack_);
    }

    vEventGroupDelete(event_group_);
}

void WakeWordDetect::Initialize(int channels, bool reference) {
    channels_ = channels;
    reference_ = reference;
    int ref_num = reference_ ? 1 : 0;

    srmodel_list_t *models = esp_srmodel_init("model");
    for (int i = 0; i < models->num; i++) {
        ESP_LOGI(TAG, "Model %d: %s", i, models->model_name[i]);
        if (strstr(models->model_name[i], ESP_WN_PREFIX) != NULL) {
            wakenet_model_ = models->model_name[i];
            auto words = esp_srmodel_get_wake_words(models, wakenet_model_);
            // split by ";" to get all wake words
            std::stringstream ss(words);
            std::string word;
            while (std::getline(ss, word, ';')) {
                wake_words_.push_back(word);
            }
        }
    }

    afe_handle_ = (esp_afe_sr_iface_t* )&ESP_AFE_SR_HANDLE;

    afe_config_t afe_config = {
[... 516 lines cut ...]
        wake_word_detect_.OnMultinetDetected([this](const std::string& multinet_word) {
            Schedule([this, &multinet_word]() {
                //TODO 这里需要优化，调整逻辑
                ESP_LOGI(TAG, "Multnet word detected: %s", multinet_word.c_str());
                if (device_state_ == kDeviceStateIdle) {
                    
                } else if (device_state_ == kDeviceStateSpeaking) {
                    AbortSpeaking(kAbortReasonWakeWordDetected);
                } else if (device_state_ == kDeviceStateActivating) {
                    SetDeviceState(kDeviceStateIdle);
                }
                // Resume detection
                wake_word_detect_.StartDetection();
            });
        });
    });
    wake_word_detect_.StartDetection();
#endif

    SetDeviceState(kDeviceStateIdle);
    esp_timer_start_periodic(clock_timer_handle_, 1000000);
}
```

### 我要实现的功能是，支持语音唤醒和命令词检测，唤醒后延时检测命令词，当没有检测到命令词时为正常的唤醒流程，发送唤醒词及唤醒音频，检测到命令词后执行本地设备控制，不发送唤醒词及音频数据到云端。这段OnMultinetDetected方法应该如何修改，搞了几天了，正常唤醒聊天没问题，当检测到命令词后，就处于一直聆听状态。

fengjiannan2010, 2025-03-03:
那我应该如何实现，有好的建议吗？物联网指令需要服务端支持吧。

78, 2025-03-03:
服务器只是支持协议，物联网的设备有哪些能力和如何控制，是在设备上提供的

closed 2025-05-20 by 78