78/xiaozhi-esp32 #204: 虽然虾哥说了iot不完善，但还是想问一下

xiaodiao216, 2025-02-21:
1.语音识别后，执行iot部分指令，为啥无论是说：“关闭客厅灯”“关闭书房灯”，小智都会执行lamp的turnoff不区分设备？
2.用gpio口控制其他设备可能就在机器人、小车方面应用多一些；智能家居使用的话，可能更多的会使用到mqtt的publish功能，请问这样的话是不是需要自己初始化一个mqtt类（使用巴法云或者其他云）来实现智能设备无线控制？

78, 2025-02-21:
2. 是的。设备上可以有多个mqtt连接的。
1. 可能是写法有误导性，应该贴出你的代码看看。

xiaodiao216, 2025-02-22:
解决了，无论书房灯还是客厅灯都执行lamp的函数应该是因为，示例程序只有一个“灯”，增加书房灯后不再混淆
1 mqtt_protocol增加另个函数
```
bool MqttProtocol::StartBafaMqttClient() {
    if (mqtt_bafa != nullptr) {
        ESP_LOGW(TAG, "BafaMqtt client already started");
        delete mqtt_bafa;
    }
    
    mqtt_bafa = Board::GetInstance().CreateMqtt();
    mqtt_bafa->SetKeepAlive(90);

    mqtt_bafa->OnDisconnected([this]() {
        ESP_LOGI(TAG, "BafaMqtt Disconnected from endpoint");
    });

    if (!mqtt_bafa->Connect(BAFA_ENDPOINT, 9501, BAFA_CLIENTID,"","")) {
        ESP_LOGE(TAG, "BafaMqtt Failed to connect to endpoint");
        if (on_network_error_ != nullptr) {
            on_network_error_(Lang::Strings::UNABLE_TO_CONNECT_TO_SERVICE);
        }
        return false;
    };

    ESP_LOGI(TAG, "BafaMqtt Connected to endpoint");
    return true;
}
```

```
bool MqttProtocol::CustomPublish(const std::string& topic, const std::string& payload) {
    if (mqtt_bafa == nullptr) {
        ESP_LOGE(TAG, "Custom_Mqtt client is not started");
        return false;
    }
    return mqtt_bafa->Publish(topic, payload);
}
```
2 iot/things文件夹增加一个bookroomlamp类，当然还要增加一些变量用（不详述）
```
Bookroomlamp() : Thing("Bookroomlamp", "书房灯"), power_(false) {
        // 定义设备的属性
        properties_.AddBooleanProperty("power", "灯是否打开", [this]() -> bool {
            return power_;
        });

        // 定义设备可以被远程执行的指令
        methods_.AddMethod("TurnOn", "打开书房灯", ParameterList(), [this](const ParameterList& parameters) {
            ESP_LOGI(TAG, "开书房灯函数");
            power_ = true;
            bafamqtt.StartBafaMqttClient();
            bafamqtt.CustomPublish("8266light", "on");
            
        });

        methods_.AddMethod("TurnOff", "关闭书房灯", ParameterList(), [this](const ParameterList& parameters) {
            ESP_LOGI(TAG, "关书房灯函数");
            power_ = false;
            bafamqtt.StartBafaMqttClient();
            bafamqtt.CustomPublish("8266light", "off");
        });
    }
};

} // namespace iot
```
3 找到自己硬件对应的主板文件 我的是boards/bread_compat_wifi/compat_wifi_board.cc，在InitializeIot() 函数中增加自己的设备
```
void InitializeIot() {
        auto& thing_manager = iot::ThingManager::GetInstance();
        thing_manager.AddThing(iot::CreateThing("Speaker"));
        thing_manager.AddThing(iot::CreateThing("Lamp"));
        thing_manager.AddThing(iot::CreateThing("Bookroomlamp"));//这个是我的新增
        thing_manager.AddThing(iot::CreateThing("Backlight"));
    }
```
4 题外：巴法云上面注册一个mqtt设备8266light，8266设备通过订阅巴法云相关主题实现控制

78, 2025-02-22:
Wow，已经可以工作了吗，可以发个视频看看效果。

你自己定义的iot类，可以写在开发板目录下。后面看能不能改进一下iot的功能，比如定义一个灯的class，但是可以创建多个出来，传入不同的参数名字，这样就能解决很多灯，但是只需要实现一次。

xiaodiao216, 2025-02-22:
好的虾哥，我研究下看看能不能一个类控制一类设备

https://github.com/user-attachments/assets/23f3a49f-646b-4f24-a127-dac4eae4911b

78, 2025-02-22:
可能要修改thing和thing manager来支持生成多个灯的实例，但是通过传入不同的参数和描述来区别。

dengjb, 2025-02-25:
请问下，将语音转为iot指令是大模型的能力吗？还是需要单独引入其他框架？

> 可能要修改thing和thing manager来支持生成多个灯的实例，但是通过传入不同的参数和描述来区别。

78, 2025-02-26:
大模型的能力，qwen-agent就是开源的实现。最近看到群里有提到mcp。

sunq945, 2025-03-03:
年前买了个esp32s3模块折腾着玩，用它作了个暖风控制，接入局域网的mqtt broker，控制暖风，然后，我经常在电脑浏览器上打开暖风，然后睡觉的时候，关了电脑之后，让小智去关暖风，结果呢，小智就做下病了，每次让她退下的时候，她就傻不啦叽的给我把暖风关了，然后我就怼她，她就给我道歉，说以后不会主动动我的设备了，在关闭之前都会询问我。然后就好了一两天吧。我还是在电脑上开暖风，让小智去关，结果，她老毛病又犯了，退出时就自动给我关了。然后我就骂她，她就说，那好吧，我把之前记录的你的偏好都清除了。我说，都清除干净！然后就正常了一阵子。有时候我说，关暖气，她也会理解为关暖风。所以，这些控制，不是精准的按照命令的每一个字去控制。所以，这个还是需要去训练她......

78, 2025-03-03:
> 年前买了个esp32s3模块折腾着玩，用它作了个暖风控制，接入局域网的mqtt [...]

换个模型，比如 qwen-max 和 deepseek 能理解关暖风和关暖气的区别？

vip-8888, 2025-03-17:
@xiaodiao216    添加巴法的 mqtt还要在mqtt_protocol.h里加入什么，， 编译出现StartBafaMqttClient()  错误，还望指教

xiaodiao216, 2025-03-17:
> [@xiaodiao216](https://github.com/xiaodiao216) 添加巴法的 mqtt还要在mqtt_protocol.h里加入什么，， 编译出现StartBafaMqttClient() 错误，还望指教

public函数定义那块加上    bool StartBafaMqttClient();

vip-8888, 2025-03-17:
@xiaodiao216   还是编译不了 有改动的文件 mqtt_protocol.h  mqtt_protocol.cc bookroomlamp.cc sample_interface.json的完整代码 抄下

vip-8888, 2025-03-17:
@xiaodiao216  
 // bafamqtt.StartBafaMqttClient();
      //  bafamqtt.CustomPublish("8266light", "on");   和off

vip-8888, 2025-03-17:
@xiaodiao216    屏蔽掉4行 就能编译，，  调用bafamqtt.StartBafaMqttClient  不行 ，好像是缺啥

vip-8888, 2025-04-02:
@78 在 iot-things里新建cc，用 esp_mqtt.h  ， Connect连接巴法云，Subscribe订阅主题 ，methods_.AddMethod里Publish 发送1 0，被控制8266网络设备也订阅主题，接收1 0控制io

ksnzh, 2025-05-12:
`mqtt_client: esp_mqtt_handle_transport_read_error: transport_read() error: errno=128`

出现这个可能是啥原因 @78