78/xiaozhi-esp32 #208: GPIO_NUM_41, GPIO_NUM_42, GPIO_NUM_47, GPIO_NUM_40, GPIO_NUM_48 undefined

jiapei100, 2025-02-21:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

1.2.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

`idf.py build` successfully.

### What is the actual behavior?

With the following error messages:

```
....../xiaozhi-esp32/main/boards/bread-compact-wifi/config.h:37:25: error: 'GPIO_NUM_41' was not declared in this scope; did you mean 'GPIO_NUM_4'?
   37 | #define DISPLAY_SDA_PIN GPIO_NUM_41
      |                         ^~~~~~~~~~~
....../xiaozhi-esp32/main/boards/bread-compact-wifi/compact_wifi_board.cc:31:27: note: in expansion of macro 'DISPLAY_SDA_PIN'
   31 |             .sda_io_num = DISPLAY_SDA_PIN,
      |                           ^~~~~~~~~~~~~~~
....../xiaozhi-esp32/main/boards/bread-compact-wifi/config.h:38:25: error: 'GPIO_NUM_42' was not declared in this scope; did you mean 'GPIO_NUM_4'?
   38 | #define DISPLAY_SCL_PIN GPIO_NUM_42
      |                         ^~~~~~~~~~~
....../xiaozhi-esp32/main/boards/bread-compact-wifi/compact_wifi_board.cc:32:27: note: in expansion of macro 'DISPLAY_SCL_PIN'
   32 |             .scl_io_num = DISPLAY_SCL_PIN,
      |                           ^~~~~~~~~~~~~~~
....../xiaozhi-esp32/main/boards/bread-compact-wifi/compact_wifi_board.cc: In constructor 'CompactWifiBoard::CompactWifiBoard()':
....../xiaozhi-esp32/main/boards/bread-compact-wifi/config.h:33:33: error: 'GPIO_NUM_47' was not declared in this scope; did you mean 'GPIO_NUM_7'?
   33 | #define TOUCH_BUTTON_GPIO       GPIO_NUM_47
      |                                 ^~~~~~~~~~~
....../xiaozhi-esp32/main/boards/bread-compact-wifi/compact_wifi_board.cc:100:23: note: in expansion of macro 'TOUCH_BUTTON_GPIO'
  100 |         touch_button_(TOUCH_BUTTON_GPIO),
      |                       ^~~~~~~~~~~~~~~~~
....../xiaozhi-esp32/main/boards/bread-compact-wifi/config.h:34:33: error: 'GPIO_NUM_40' was not declared in this scope; did you mean 'GPIO_NUM_4'?
   34 | #define VOLUME_UP_BUTTON_GPIO   GPIO_NUM_40
      |                                 ^~~~~~~~~~~
....../xiaozhi-esp32/main/boards/bread-compact-wifi/compact_wifi_board.cc:101:27: note: in expansion of macro 'VOLUME_UP_BUTTON_GPIO'
  101 |         volume_up_button_(VOLUME_UP_BUTTON_GPIO),
      |                           ^~~~~~~~~~~~~~~~~~~~~
....../xiaozhi-esp32/main/boards/bread-compact-wifi/compact_wifi_board.cc: In member function 'virtual Led* CompactWifiBoard::GetLed()':
....../xiaozhi-esp32/main/boards/bread-compact-wifi/config.h:31:33: error: 'GPIO_NUM_48' was not declared in this scope; did you mean 'GPIO_NUM_8'?
   31 | #define BUILTIN_LED_GPIO        GPIO_NUM_48
      |                                 ^~~~~~~~~~~
....../xiaozhi-esp32/main/boards/bread-compact-wifi/compact_wifi_board.cc:109:30: note: in expansion of macro 'BUILTIN_LED_GPIO'
  109 |         static SingleLed led(BUILTIN_LED_GPIO);
      |                              ^~~~~~~~~~~~~~~~
```

### Steps to reproduce.

Nothing particular, just build

### Build or installation Logs.

```plain
N/A
```

csh-cret, 2025-02-21:
是不是应该在idf.py build之前进行设备的选择：idf.py set-target esp32s3

78, 2025-02-21:
操作问题。

luckyxiangfeng, 2025-03-21:
我也是报这个错误，你是这么解决的

FlyerChao, 2025-03-24:
E:/project/xiaozhi/xiaozhi-esp32/main/boards/lichuang-dev/lichuang_dev_board.cc:66:30: error: 'GPIO_NUM_40' was not declared in this scope; did you mean 'GPIO_NUM_4'?
   66 |         buscfg.mosi_io_num = GPIO_NUM_40;
      |                              ^~~~~~~~~~~
      |                              GPIO_NUM_4
E:/project/xiaozhi/xiaozhi-esp32/main/boards/lichuang-dev/lichuang_dev_board.cc:68:30: error: 'GPIO_NUM_41' was not declared in this scope; did you mean 'GPIO_NUM_4'?
   68 |         buscfg.sclk_io_num = GPIO_NUM_41;
      |                              ^~~~~~~~~~~
      |                              GPIO_NUM_4
In file included from E:/project/xiaozhi/xiaozhi-esp32/main/boards/lichuang-dev/lichuang_dev_board.cc:6:
E:/project/xiaozhi/xiaozhi-esp32/main/boards/lichuang-dev/lichuang_dev_board.cc: In member function 'virtual AudioCodec* LichuangDevBoard::GetAudioCodec()':
E:/project/xiaozhi/xiaozhi-esp32/main/boards/lichuang-dev/config.h:15:29: error: 'GPIO_NUM_45' was not declared in this scope; did you mean 'GPIO_NUM_5'?
   15 | #define AUDIO_I2S_GPIO_DOUT GPIO_NUM_45
      |                             ^~~~~~~~~~~
E:/project/xiaozhi/xiaozhi-esp32/main/boards/lichuang-dev/lichuang_dev_board.cc:158:13: note: in expansion of macro 'AUDIO_I2S_GPIO_DOUT'
  158 |             AUDIO_I2S_GPIO_DOUT,
      |             ^~~~~~~~~~~~~~~~~~~
E:/project/xiaozhi/xiaozhi-esp32/main/boards/lichuang-dev/lichuang_dev_board.cc: In member function 'virtual Backlight* LichuangDevBoard::GetBacklight()':
E:/project/xiaozhi/xiaozhi-esp32/main/boards/lichuang-dev/config.h:37:31: error: 'GPIO_NUM_42' was not declared in this scope; did you mean 'GPIO_NUM_4'?
   37 | #define DISPLAY_BACKLIGHT_PIN GPIO_NUM_42
      |                               ^~~~~~~~~~~
E:/project/xiaozhi/xiaozhi-esp32/main/boards/lichuang-dev/lichuang_dev_board.cc:172:39: note: in expansion of macro 'DISPLAY_BACKLIGHT_PIN'
  172 |         static PwmBacklight backlight(DISPLAY_BACKLIGHT_PIN, DISPLAY_BACKLIGHT_OUTPUT_INVERT);
      |                                       ^~~~~~~~~~~~~~~~~~~~~

选择了lichuang的S3的device，编译还是报错。

78, 2025-03-24:
看 @csh-cret 的答案。