78/xiaozhi-esp32 #211: 可以增加对sh1106芯片的OLED屏幕的适配吗？

Rueded, 2025-02-21:
### Is your feature request related to a problem?

使用的1.3寸 128*64的OLED屏幕不适配，显示花屏，后来才发现代码仅支持SSD1306芯片的适配

### Describe the solution you'd like.

建议增加对sh1106的适配，也可以使用u8g2让更多屏幕兼容

### Describe alternatives you've considered.

或者能给出其他解决方案吗？我能修改ssd1306.cc的代码让其对sh1106屏幕的适配，但是我不懂该怎么放进 .bin文件里边，有大佬教一下吗？

78, 2025-02-21:
需要驱动，目前没有找到乐鑫官方提供的驱动呢。

Rueded, 2025-02-21:
这位大佬有做关于屏幕的驱动，很多型号都适配，也支持用于esp-idf，或许可行。https://github.com/lexus2k/lcdgfx

> 需要驱动，目前没有找到乐鑫官方提供的驱动呢。

78, 2025-02-21:
需要支持esp idf才行？

sayhello00, 2025-03-01:
> ### Is your feature request related to a problem?
> 使用的1.3寸 128*64的OLED屏幕不适配，显示花屏，后来才发现代码仅支持SSD1306芯片的适配
>
> ### Describe the solution you'd like.
> 建议增加对sh1106的适配，也可以使用u8g2让更多屏幕兼容
>
> ### Describe [...]

>###其他解决方案
>借助第三方库，作者仓库地址：[https://github.com/TNY-Robotics/sh1106-esp-idf](url)

>1. 在ESPIDF终端输入命令`idf.py add-dependency "tny-robotics/sh1106-esp-idf^1.0.0"`，构建完managed_components目录会多出一个名为tny-robotics__sh1106-esp-idf的文件夹
>2. 参考main文件夹中的示例修改ssd1306_display.cc文件(复制一个出来改包括头文件)
>3. 打开boards文件夹内的xx_board.cc文件(取决于你的主板类型)，修改GetDisplay函数，Ssd1306Display换成Sh1106Display，更换sh1106的头文件
>4. 打开main文件夹的CMakeList.txt，添加或替换一个"display/sh1106_display.cc"
>5. 构建，排查报错，烧录

zhengruiw02, 2025-03-04:
https://github.com/78/xiaozhi-esp32/compare/main...zhengruiw02:xiaozhi-esp32:dev_boards

之前手上有个SH1106的128*64的OLED屏幕，写过一个驱动，可以试一下这个。

kiepphongba, 2025-08-20:
你能给我运行 oled 0.96 的 esp32 s3 n16r8 版本的固件吗？