78/xiaozhi-esp32 #212: 立创·实战派ESP32-C3开发板 编译选项

bright8192, 2025-02-21:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.2.1

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

固件编译设置选项使用 能否给出详细的说明，编译成功刷入无法正常工作。硬件是 https://oshwhub.com/li-chuang-kai-fa-ban/xd-esp32c3-aiot-v1_3_2

### What is the actual behavior?

编译成功刷入后无法正常工作

### Steps to reproduce.

[image]
[image]
[image]

### Build or installation Logs.

```plain

```

78, 2025-02-21:
Try
`python scripts/release.py lichuang-c3-dev`

heycms, 2025-02-25:
Partition Table  也要改一下 partitions_8M.csv

bright8192, 2025-03-03:
重新编译字体调小就好了。

bright8192, 2025-03-03:
@78 谢谢 虾哥。

closed 2025-03-03 by bright8192