78/xiaozhi-esp32 #214: 提交一个 esp-sparkbot 小坦克屏幕的反色问题

dancemanUK, 2025-02-21:
在 esp_sparkbot_board.cc 代码里，第114行

        esp_lcd_panel_invert_color(panel, false);   

这里的反色为： false; 效果如下图

[image]

将 esp_lcd_panel_invert_color(panel, true);   
才能正常显示

78, 2025-02-21:
OK. Next time, please submit a pull request to address this issue.

78, 2025-02-21:
Reopen again.

尝试把 rgb_ele_order 改为 BGR，能否修复你的问题。

dancemanUK, 2025-02-23:
> Reopen again.
> 
> 尝试把 rgb_ele_order 改为 BGR，能否修复你的问题。

虾哥，这个是ST7789的240X240的屏。LCD_RGB_ELEMENT_ORDER_RGB; 这里没有问题，
只要esp_lcd_panel_invert_color(panel, true); 就能正常白底黑色，和显示黄色的图标。

原代码提交者，可能是要黑底白字，又不想去修改。lcd_display.cc 里的LVGL代码，就直接改 esp_lcd_panel_invert_color(panel, false);

lifewinnerP11, 2025-03-08:
> > Reopen again.
> > 尝试把 rgb_ele_order 改为 BGR，能否修复你的问题。
> 
> 虾哥，这个是ST7789的240X240的屏。LCD_RGB_ELEMENT_ORDER_RGB; 这里没有问题， 只要esp_lcd_panel_invert_color(panel, true); 就能正常白底黑色，和显示黄色的图标。
> 
> 原代码提交者，可能是要黑底白字，又不想去修改。lcd_display.cc 里的LVGL代码，就直接改 esp_lcd_panel_invert_color(panel, false);

我也想实现黑底白字的设置，请问一下大神知道该如何修改吗？小白谢谢

closed 2025-03-21 by 78