78/xiaozhi-esp32 #216: 单mic机型实现aec（回声消除）的可行性

ACI0419, 2025-02-21:
### Is your feature request related to a problem?

我之前在群里面反馈了1.2.1版本 面包板 s3 wakenet aec（语音唤醒回声消除）不生效的问题。

### Describe the solution you'd like.

我仔细研究了一下代码，发现这个项目目前只有双mic的机型支持这个功能。如果我没看错的话，aec ref似乎是使用双mic的其中一个，另一个作为输入。但是根据我查的资料，aec ref可以直接从扬声器i2s回采，不需要通过麦克风也能获取ref。也就是说，单mic也可以做回声消除。希望后续可以新增这个功能

### Additional context.

我本来想直接在这个项目上验证我的想法，但是我目前还不是很熟悉这个项目（感觉这个项目二次开发的文档真挺少的，注释也不多）。花了一整天的时间才摸清项目结构定位到问题 XP

78, 2025-02-21:
从硬件上做回采效果会更好，功放也会改变音频数据，所以拿到功放之后的数据效果应该是最好的。
之前有做过实验，基于esp-box-3，大概步骤是：

1. 把opus complexity设为0释放出更多CPU。
2. 使用BoxAudioCodec，开启reference
3. 把audio processing的aec打开
4. 最后，还要修改speaking的时候，注释掉audio_processor_.Stop。

 差不多这样，太久了都忘了。效果：https://www.bilibili.com/video/BV1vx1AYQEhZ/

ACI0419, 2025-02-22:
希望后续版本的menuconfig可以把wakeupNet和AudioProcess这两个分开，然后wakeupNet具体使用mic回采还是扬声器i2s回采也可以选择，这样没有双mic的机型也能用上aec

78, 2025-02-22:
是可以的，欢迎提交代码！

ACI0419, 2025-02-26:
不对啊，我忽然意识到，双mic的机器，两个mic获取的都是环境中的声音，都包含用户说话的声音。使用mic输入作为ref,那不是把用户的声音也降没了？除非作为ref的mic离扬声器很近，信噪比够低。

78, 2025-02-26:
有一路mic是连接spk的，叫回采。

ACI0419, 2025-02-27:
原来是硬件上连接了啊。我没看原理图XD

78, 2025-04-01:
最新版本已支持立创开发板S3、ESP Box、Korvo开发板的AEC和语音打断。关闭本issue。

closed 2025-04-01 by 78

hyansuper, 2025-04-07:
> 原来是硬件上连接了啊。我没看原理图XD

@ACI0419 大佬实现了吗？我觉得 单 mic 因该也可以，只要能实现回采。
像[立创开发板](https://lckfb.com/project/detail/lckfb-esp32-s3-va?param=baseInfo&collection=71ac02fcf595444eb5dbbc45ae4ffda8)用了个 ADC es7210， 同时接入mic 和speaker 的信号，（软件或硬件？）的算法上就能消除 mic 中的 speaker 声。如果是单 mic 就在配置里也设置好就行

78, 2025-04-07:
单mic是可以的，主要是要把ns4150b之后的模拟信号，接到es7210的其中一路mic上，传回mcu。

rx-union, 2025-07-09:
ES8388是否也可以呢？7210+8311毕竟使用了两个芯片，对于功耗和成本都会有增加

78, 2025-07-10:
> ES8388是否也可以呢？7210+8311毕竟使用了两个芯片，对于功耗和成本都会有增加

可以的

hyansuper, 2025-07-10:
> > ES8388是否也可以呢？7210+8311毕竟使用了两个芯片，对于功耗和成本都会有增加
> 
> 可以的

😧单一个ES8388只有一路输入，连接mic。喇叭声怎么回采？
请问目前源码中哪个设备的配置只用es8388就实现了该功能？

78, 2025-07-10:
8388应该是2路输入+2路输出，你可以搜一下代码