78/xiaozhi-esp32 #218: 小智聊着聊着 就变成待命状态了 立创S3

jia110, 2025-02-22:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

1.2.1

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

小智聊着聊着   几句话之后  就变成待命状态了  

### What is the actual behavior?

小智聊着聊着 就变成待命状态了  

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

78, 2025-02-22:
使用什么模型吗？

jia110, 2025-02-22:
使用了DeepSeek V3      不知道跟让他唱歌有没有关系  让它唱歌 这种情况 出现了两次  歌曲还没播放 就待命了

78, 2025-02-22:
deepseek没有qwen那么完善，还需要优化。

jia110, 2025-02-23:
原来如此

closed 2025-02-23 by jia110

78, 2025-02-23:
初步推断应该是DeepSeek输出了格式错误的JSON内容，服务端解析错误抛出异常，关闭了对话。