78/xiaozhi-esp32 #220: 新增对微雪1.56开发版的支持

tcking, 2025-02-22:
### Is your feature request related to a problem?

Hi，从工程的commit中看有添加对 微雪电子-Touch-LCD-1.46 开发版的支持，不过从readme中没有看到新增开发板，不知是否已经支持完全，我使用最新的master代码编译并烧录到开发版，上电之后出行不停的reboot，日志如下：
```
Hi，我用最新的master代码 +  微雪的1.46的开发板（https://www.waveshare.net/wiki/ESP32-S3-Touch-LCD-1.46），上电之后会出现不断的reboot，日志如下：

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379eb1
--- 0x40379eb1: esp_restart_noos at /Users/didi/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:160

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x564
load:0x403c8700,len:0xc38
--- 0x403c8700: _stext at ??:?

load:0x403cb700,len:0x2ea0
entry 0x403c8908
--- 0x403c8908: call_start_cpu0 at /Users/didi/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25

I (34) octal_psram: vendor id    : 0x0d (AP)
I (34) octal_psram: dev id       : 0x02 (generation 3)
I (34) octal_psram: density      : 0x03 (64 Mbit)
I (36) octal_psram: good-die     : 0x01 (Pass)
I (40) octal_psram: Latency      : 0x01 (Fixed)
I (44) octal_psram: VCC          : 0x01 (3V)
I (48) octal_psram: SRF          : 0x01 (Fast Refresh)
I (53) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (58) octal_psram: BurstLen     : 0x01 (32 Byte)
I (62) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (67) octal_psram: DriveStrength: 0x00 (1/1)
I (72) MSPI Timing: PSRAM timing tuning index: 5
I (76) esp_psram: Found 8MB PSRAM device
I (79) esp_psram: Speed: 80MHz
I (82) cpu_start: Multicore app
I (97) cpu_start: Pro cpu start user code
I (97) cpu_start: cpu freq: 240000000 Hz
I (97) app_init: Application information:
I (97) app_init: Project name:     xiaozhi
I (100) app_init: App version:      1.2.2
I (104) app_init: Compile time:     Feb 22 2025 17:40:11
I (109) app_init: ELF file SHA256:  d4000251c...
I (114) app_init: ESP-IDF:          v5.5-dev-1655-gc5865270b5
I (119) efuse_init: Min chip rev:     v0.0
I (123) efuse_init: Max chip rev:     v0.99 
I (127) efuse_init: Chip rev:         v0.2
I (131) heap_init: Initializing. RAM available for dynamic allocation:
I (137) heap_init: At 3FCAEB10 len 0003AC00 (235 KiB): RAM
I (142) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (147) heap_init: At 600FE01C len 00001FBC (7 KiB): RTCRAM
I (153) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (160) spi_flash: detected chip: winbond
I (163) spi_flash: flash io: qio
I (167) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (172) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (179) main_task: Started on CPU0
I (189) esp_psram: Reserving pool of 48K of internal memory for DMA/internal allocations
I (189) main_task: Calling app_main()
I (209) BackgroundTask: background_task started
I (209) Board: UUID: a2d90a77-fd88-4453-b658-866e5606344e
I (209) button: IoT Button Version: 3.5.0
I (219) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (229) sh8601: LCD panel create success, version: 1.0.2
E (459) sh8601: swap_xy is not supported by this panel
I (529) LcdDisplay: Turning display on
I (529) LcdDisplay: Initialize LVGL library
I (529) LcdDisplay: Initialize LVGL port
[... 90 lines cut ...]
E (1089) i2c.master: I2C transaction unexpected nack detected
E (1099) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (1099) i2c.master: i2c_master_multi_buffer_transmit(1193): I2C transaction failed
E (1109) I2C_If: Fail to write to dev 30
E (1109) ES8311: Open fail

assert failed: Es8311AudioCodec::Es8311AudioCodec(void*, i2c_port_t, int, int, gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t, uint8_t, bool) es8311_audio_codec.cc:48 (codec_i

Backtrace: 0x40379f71:0x3fcb2420 0x40384969:0x3fcb2440 0x4038d2ad:0x3fcb2460 0x4200da36:0x3fcb2580 0x4201a7b6:0x3fcb2600 0x42012892:0x3fcb2640 0x42012fd6:0x3fcb2660 0x420168fb:0x3fcb2870 0x421ac468:0x3fcb28a0 0x4038545d:0x3fcb28d0
--- 0x40379f71: panic_abort at /Users/didi/esp/esp-idf/components/esp_system/panic.c:454
0x40384969: esp_system_abort at /Users/didi/esp/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x4038d2ad: __assert_func at /Users/didi/esp/esp-idf/components/newlib/src/assert.c:80
0x4200da36: Es8311AudioCodec::Es8311AudioCodec(void*, i2c_port_t, int, int, gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t, unsigned char, bool) at /Users/didi/workspace/esp32/xiaozhi-esp32/main/audio_codecs/es8311_audio_codec.cc:48 (discriminator 1)
0x4201a7b6: waveshare_amoled_1_8::GetAudioCodec() at /Users/didi/workspace/esp32/xiaozhi-esp32/main/boards/esp32-s3-touch-amoled-1.8/esp32-s3-touch-amoled-1.8.cc:170 (discriminator 1)
0x42012892: Application::SetDeviceState(DeviceState) at /Users/didi/workspace/esp32/xiaozhi-esp32/main/application.cc:669
0x42012fd6: Application::Start() at /Users/didi/workspace/esp32/xiaozhi-esp32/main/application.cc:289
0x420168fb: app_main at /Users/didi/workspace/esp32/xiaozhi-esp32/main/main.cc:28 (discriminator 1)
0x421ac468: main_task at /Users/didi/esp/esp-idf/components/freertos/app_startup.c:208
0x4038545d: vPortTaskWrapper at /Users/didi/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: d4000251c

Rebooting...
``` 
```

chao, 2025-02-23:
很奇怪，手头有一块去年11月份买的1.46板子，直接编译烧录没有任何问题。同样编译出来固件烧录到2月份买的1.46上也一直在重启。看了下日志：

```
]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0 
E (293) i2c.master: I2C transaction unexpected nack detected
E (293) i2c.master: s_i2c_synchronous_transaction(888): I2C transaction failed
E (303) i2c.master: i2c_master_transmit(1116): I2C transaction failed
E (313) tca9554: write_direction_reg(127): Write direction reg failed
E (323) tca9554: reset(142): Write dir reg failed
E (323) tca9554: esp_io_expander_new_i2c_tca9554(81): Reset failed
E (333) waveshare_lcd_1_46: TCA9554 create returned error
E (333) io_expander: esp_io_expander_set_dir(35): Invalid handle
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x42019e0a
--- 0x42019e0a: CustomBoard::i2c_dev_tca9554_init() at /Users/zhouchao/Projects/esp32/xiaozhi-esp32-main/main/boards/esp32-s3-touch-lcd-1.46/esp32-s3-touch-lcd-1.46.cc:101 (discriminator 1)

file: "./main/boards/esp32-s3-touch-lcd-1.46/esp32-s3-touch-lcd-1.46.cc" line 101
func: void CustomBoard::i2c_dev_tca9554_init()
expression: ret

abort() was called at PC 0x4038477f on core 0
.--- 0x4038477f: _esp_error_check_failed at /Users/zhouchao/esp/v5.3.2/esp-idf/components/esp_system/esp_err.c:49

Backtrace: 0x40379fba:0x3fcb24d0 0x40384789:0x3fcb24f0 0x4038cfd9:0x3fcb2510 0x4038477f:0x3fcb2580 0x42019e0a:0x3fcb25b0 0x4201a1a3:0x3fcb25d0 0x4201a201:0x3fcb25f0 0x4200d19d:0x3fcb2610 0x42012072:0x3fcb2630 0x42015faf:0x3fcb2830 0x421a75a3:0x3fcb2860 0x40385339:0x3fcb2890
..--- 0x40379fba: panic_abort at /Users/zhouchao/esp/v5.3.2/esp-idf/components/esp_system/panic.c:463
0x40384789: esp_system_abort at /Users/zhouchao/esp/v5.3.2/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x4038cfd9: abort at /Users/zhouchao/esp/v5.3.2/esp-idf/components/newlib/abort.c:38
0x4038477f: _esp_error_check_failed at /Users/zhouchao/esp/v5.3.2/esp-idf/components/esp_system/esp_err.c:49
0x42019e0a: CustomBoard::i2c_dev_tca9554_init() at /Users/zhouchao/Projects/esp32/xiaozhi-esp32-main/main/boards/esp32-s3-touch-lcd-1.46/esp32-s3-touch-lcd-1.46.cc:101 (discriminator 1)
0x4201a1a3: CustomBoard::CustomBoard() at /Users/zhouchao/Projects/esp32/xiaozhi-esp32-main/main/boards/esp32-s3-touch-lcd-1.46/esp32-s3-touch-lcd-1.46.cc:178
0x4201a201: create_board() at /Users/zhouchao/Projects/esp32/xiaozhi-esp32-main/main/boards/esp32-s3-touch-lcd-1.46/esp32-s3-touch-lcd-1.46.cc:199 (discriminator 1)
0x4200d19d: Board::GetInstance() at /Users/zhouchao/Projects/esp32/xiaozhi-esp32-main/main/boards/common/board.h:30 (discriminator 1)
0x42012072: Application::Start() at /Users/zhouchao/Projects/esp32/xiaozhi-esp32-main/main/application.cc:258
0x42015faf: app_main at /Users/zhouchao/Projects/esp32/xiaozhi-esp32-main/main/main.cc:28 (discriminator 1)
0x421a75a3: main_task at /Users/zhouchao/esp/v5.3.2/esp-idf/components/freertos/app_startup.c:208
0x40385339: vPortTaskWrapper at /Users/zhouchao/esp/v5.3.2/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

是初始化tca9554的时候报错了。把i2c_dev_tca9554_init();从esp32-s3-touch-lcd-1.46.cc里注释了以后再编译就没问题。

78, 2025-02-23:
@Sail-211010 麻烦看看。 #159

chao, 2025-02-23:
我这个怀疑是电路有问题，改了下InitializeI2c的代码，把I2C的地址都列出来看了一下。只有0x51, 0x6B, 0x7E，没有TCA9554的0x20

```
void InitializeI2c() {
        // Initialize I2C peripheral
        i2c_master_bus_config_t i2c_bus_cfg = {
            .i2c_port = (i2c_port_t)0,
            .sda_io_num = I2C_SDA_IO,
            .scl_io_num = I2C_SCL_IO,
            .clk_source = I2C_CLK_SRC_DEFAULT,
        };
        ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_bus_cfg, &i2c_bus_));

        // Scan for I2C devices
        uint8_t address;
        ESP_LOGI(TAG, "Scanning I2C bus...");
        for (address = 1; address < 127; address++) {
            i2c_master_bus_handle_t cmd = i2c_bus_;
            esp_err_t ret;
            ret = i2c_master_probe(cmd, address, -1);
            if (ret == ESP_OK) {
                ESP_LOGI(TAG, "Found I2C device at address 0x%02x", address);
            }
        }
    }
```

看了下原理图
[image]

这个地址是0x20。手头没设备，明天到公司测一下再同步信息。

chao, 2025-02-24:
确认了是TCA9554PWR模组有问题。跟程序无关。

Sail-211010, 2025-02-24:
嗯  有相关问题可以联系售后

tcking, 2025-03-01:
感觉各位的支持，checkout最新的master代码可以正常运行在1.46上了，就是这个板子的音量太小了，我看看能否将声音通过蓝牙音响播放出来

closed 2025-03-01 by tcking