78/xiaozhi-esp32 #221: Add memory to enable personalization and long term memory for all interactions

taranjeet, 2025-02-23:
### Is your feature request related to a problem?

Hi, great project, but it forgets everything.

### Describe the solution you'd like.

I would love to add a OSS memory layer (like https://github.com/mem0ai/mem0) to it

Disclaimer: I work at Mem0

78, 2025-02-23:
Hello! I have been informed of this project. The work is extremely impressive. May I implement this long-term memory within 1000 to 2000 tokens? Personal prompts are rather expensive in real-time chat.

taranjeet, 2025-02-23:
Hey thanks.
Yes memories are generally 50-100 tokens.
Would this be suitable to you?