78/xiaozhi-esp32 #223: 用idf编译完最新代码之后，固件烧录进去用不了

sbb520, 2025-02-23:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.2.1-14-g20d52de

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

根据教程：https://icnynnzcwou8.feishu.cn/wiki/JEYDwTTALi5s2zkGlFGcDiRknXf
使用最新版本的代码，在idf编译器去到目录下面，依次使用 idf.py set-target esp32s3，idf.py build，成功编译出来xiaozhi.bin文件，希望成功烧录进去面包板并可以进入到初始的配网模式。

### What is the actual behavior?

烧录出来的固件只有2.6mb大小，烧录进去面包板没有任何的反应和屏幕显示，但是如果用最新固件release：https://github.com/78/xiaozhi-esp32/releases，烧录进去面包板是可以正常使用的（下载release的固件有3.6mb大小）。怀疑是初始代码配置有问题，或者可能是小弟我的配置不对？希望能多增加一点编译代码或者修改配置的详细教程。

### Steps to reproduce.

1. git clone最新代码
2. idf.py set-target esp32s3
3. idf.py build
4. 烧录进去面包板esp32 S3
...

### Build or installation Logs.

```plain

```

KeysPAN0114, 2025-02-24:
这...好歹看一下ESP-IDF的入门教程吧。编译的固件要用idf.py -p COMX flash烧录(COMX的X在设备管理器看)，编译完的xiaozhi.bin只是软件包，其他包都没烧录，要做成一个包用官方工具合并成一个。

Rx782-Fss, 2025-02-24:
要不 就直接idf.py flash

sunq945, 2025-03-03:
编译出来一共有5个bin文件：    
    bootloader.bin （bootloader）
    ota_data_initial.bin （ota初始数据）
    partition-table.bin （分区表数据）
    srmodels.bin 
    xiaozhi.bin （app，可执行程序）
这些文件在build目录下或它的子目录下，在命令行中进入build目录之后，可以用esptool.py 的一个整合命令把这些bin档进行合并：
esptool.py --chip esp32s3 merge_bin -o merged-flash.bin --flash_mode dio --flash_size 16MB 0x0 bootloader/bootloader.bin 0x500000 xiaozhi.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin 0x10000 srmodels/srmodels.bin

以上命令应该根据自己的情况进行调整 ：
1、--flash_size 16MB ：是指定你的esp32模块的flash大小 ；
2、0x0 bootloader/bootloader.bin  ：0x0是bootloader.bin的地址，这个地址固定；
3、0x500000 xiaozhi.bin ：0x500000这是我自定义的分区表划分的地址；
4、0x8000 partition_table/partition-table.bin ：0x8000 是分区表数据存放的地址，是官方的分区表规定的，一般不需要改；
5、 0xd000 ota_data_initial.bin ：0xd000  地址是官方的分区表规定的，一般不需要改；
6、 0x10000 srmodels/srmodels.bin ：0x10000 地址是官方的分区表规定的，一般不需要改。

以上bin档的地址都在分区表上（倒数第二列），比如我的分区表（partitions.csv文件）如下：
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,    0x4000,
otadata,  data, ota,     0xd000,    0x2000,
phy_init, data, phy,     0xf000,    0x1000,
model,    data, spiffs,  0x10000,   0xF0000,
storage,  data, fat,     0x100000,  4M,
ota_0,    app,  ota_0,   0x500000,  4M,
ota_1,    app,  ota_1,   0x900000,  4M,

你可以根据你自己的分区表来修改bin档合并命令中的参数。
合并命令中 -o merged-flash.bin 意思是合并完成之后把整合文件重命名为merged-flash.bin。因为命令就是在build目录下执行的，所以这个merged-flash.bin最后生成也是在build目录下。一般会是3MB以上的文件。