78/xiaozhi-esp32 #232: 求教虾哥，当前iot代码是不是缺乏小车控制部分？

xufengnian, 2025-02-25:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

1.2.1

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

求教虾哥，我是esp-sparkbot设备，根据当前开源的[server](https://github.com/xinnan-tech/xiaozhi-esp32-server)代码，修改config.yaml，改成了自己的LLM、TTS，可以正常识别语音和对话，

但是无法像官方固件那样，通过语音来控制坦克前进、后退、灯光秀，语音识别的结果（比如“连接小车”、“坦克控制”、“向前走”等）都被LLM直接处理了，不管是gpt-4o、deepseek-r1都是一样的，

我知道服务端需要增加逻辑，识别特定指令。

**我的问题：是不是客户端还要自己写控制小车的逻辑?**

[image]

### What is the actual behavior?

如题

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

xufengnian, 2025-02-26:
客户端不缺任何代码，服务端可以补全，感谢虾哥开源

closed 2025-02-26 by xufengnian

poppinlr, 2025-03-25:
怎么补全?我也遇到了相同的疑问