78/xiaozhi-esp32 #234: MacOS上变异报错

sexiong306, 2025-02-26:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

9116dd6c07

### Operating System used.

macOS

### How did you build your project?

VS Code IDE/Cursor

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

目前还没有连接固件，在mac上搭建好esp的环境，命令行里执行
idf.py build
esp版本为
idf.py --version
ESP-IDF v5.3.2-dirty
-- Building ESP-IDF components for target esp32

### What is the actual behavior?

看起来是78__xiaozhi-fonts的问题
提示错误
`/Users/pudgeli/Documents/git/xiaozhi-esp32/managed_components/78__xiaozhi-fonts/src/font_emoji_64.c: In function 'font_emoji_64_init':
/Users/pudgeli/Documents/git/xiaozhi-esp32/managed_components/78__xiaozhi-fonts/src/font_emoji_64.c:68:16: error: implicit declaration of function 'lv_imgfont_create'; did you mean 'lv_binfont_create'? [-Werror=implicit-function-declaration]
   68 |         font = lv_imgfont_create(64, get_imgfont_path, NULL);
      |                ^~~~~~~~~~~~~~~~~
      |                lv_binfont_create
/Users/pudgeli/Documents/git/xiaozhi-esp32/managed_components/78__xiaozhi-fonts/src/font_emoji_64.c:68:14: warning: assignment to 'lv_font_t *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
   68 |         font = lv_imgfont_create(64, get_imgfont_path, NULL);
      |              ^
cc1: some warnings being treated as errors

/Users/pudgeli/Documents/git/xiaozhi-esp32/managed_components/78__xiaozhi-fonts/src/font_puhui_30_4.c:376464:4: error: #error "Too large font or glyphs in FONT_PUHUI_30_4. Enable LV_FONT_FMT_TXT_LARGE in lv_conf.h")
376464 | #  error "Too large font or glyphs in FONT_PUHUI_30_4. Enable LV_FONT_FMT_TXT_LARGE in lv_conf.h")`

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

KeysPAN0114, 2025-02-26:
> -- Building ESP-IDF components for target esp32
这不就看出问题了吗，target跟买的套件模组型号对应吗？不对应有menuconfig修改吗？

sexiong306, 2025-02-27:
> > -- Building ESP-IDF components for target esp32
> > 这不就看出问题了吗，target跟买的套件模组型号对应吗？不对应有menuconfig修改吗？

我还没接模组

shockley6668, 2025-04-23:
我也遇到了同样的问题，请问有解决方法吗