78/xiaozhi-esp32 #238: 虾哥miniC3连接本地部署的xiaozhi-esp32-server服务端接收音频数据的时候出现崩溃重启

manicwy, 2025-02-26:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.3.0 Mini-C3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

服务端端语音合成采用豆包TTS，按理说应该能全部正常反回播放

### What is the actual behavior?

在播放完第一句话后，miniC3终端崩溃重启或者卡死不动，大概率崩溃重启

[errlog.txt](https://github.com/user-attachments/files/18988455/errlog.txt)

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

manicwy, 2025-02-26:
将服务端TTS更换为EdgeTTS或者是豆包TTS，客户端在播放音频的时候同样会崩溃重启 ，除了第一句话能正常说，第二句开始大概率崩溃

78, 2025-02-26:
应该是服务器端没有做流控，没有PSRAM的C3设备内存无法存储那么多的音频数据，可以考虑内存不足时丢弃，或者在放入队列前delay一下接收任务。

xinnan-tech, 2025-03-01:
@manicwy 你好，重新拉取最新的代码试试，我们做了第一版本的流控方案