78/xiaozhi-esp32 #242: p3音效都无法播放

JohnnieFucker, 2025-02-28:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.3.0

### Operating System used.

macOS

### How did you build your project?

VS Code IDE/Cursor

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

正常播放各类p3文件音效

### What is the actual behavior?

可以正常对话，但是没有播放各种p3文件的音效，有一个W (216689) Application: Server sample rate 16000 does not match device output sample rate 24000, resampling may cause distortion

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

Kevincoooool, 2025-02-28:
P3音频转换参考：https://icnynnzcwou8.feishu.cn/wiki/I5Mfw6USWivlRYk5yzzc3nC0nCg

JohnnieFucker, 2025-02-28:
我是值的项目原有音效也无法播放

JohnnieFucker, 2025-03-07:
我发现application.cc 的OnWakeWordDetected 里面如果调用 Alert就不能播放 但是在wifi_boards.cc 里面 application.Alert 就可以正常播放，是因为OnWakeWordDetected的处理是异步的原因么？

78, 2025-03-07:
可能是后面的设置设备状态为 Listening 的时候ResetDecoder，清空了音频缓冲区导致的。
你试试把那一行改为 SetDeviceState(kDeviceStateIdle);

JohnnieFucker, 2025-03-19:
那有什么办法呢？我试过SetDeviceState(kDeviceStateIdle);不行

78, 2025-03-19:
Alert不可以放在Schedule内部。

closed 2025-03-25 by JohnnieFucker