78/xiaozhi-esp32 #245: 求助如何把屏幕设置成黑底白字

owcowc, 2025-02-28:
大神们好，技术小白，不太懂，跟着教程在做。
使用的是正点原子的esp32s3开发板，2.4寸的屏幕，成功点亮后屏幕上是白色底色黑颜色的字，还有一个表情。感觉白色在暗光下太晃眼，请问修改哪里可以把屏幕的显示变成黑色底白色字呢，谢谢！

derong218, 2025-03-05:
试试将esp_lcd_panel_invert_color的第二个参数改一下。可以翻转颜色。看看能不能满足你的要求。

owcowc, 2025-03-05:
> 试试将esp_lcd_panel_invert_color的第二个参数改一下。可以翻转颜色。看看能不能满足你的要求。

谢谢，成功了！
但是表情变成小窦尔敦了，也不错哈哈，谢谢

lifewinnerP11, 2025-03-08:
hello
我也是想改黑底白字，请问“esp_lcd_panel_invert_color的第二个参数改一下”  大神说的这个参数  是在哪里改呢  找了半天没找到
能不能帮帮小白 稍微说的详细一点 谢谢 谢谢

78, 2025-03-08:
> hello 我也是想改黑底白字，请问“esp_lcd_panel_invert_color的第二个参数改一下” 大神说的这个参数 是在哪里改呢 找了半天没找到 能不能帮帮小白 稍微说的详细一点 谢谢 谢谢

取决于你是什么开发板，没有的话，要自己加，在初始化Display的地方

owcowc, 2025-03-08:
> hello 我也是想改黑底白字，请问“esp_lcd_panel_invert_color的第二个参数改一下” 大神说的这个参数 是在哪里改呢 找了半天没找到 能不能帮帮小白 稍微说的详细一点 谢谢 谢谢

在main/boards/ 里
找到你的板子型号 修改那里边的.CC文件就行