78/xiaozhi-esp32 #246: 增加人脸识别，看到熟人主动打招呼。

SindenDev, 2025-02-28:


78, 2025-02-28:
感兴趣的可以自行实现，原理应该跟语音唤醒一样。

aonoa, 2025-02-28:
这个得板子跑得动才行，或者把视频帧发到后端去处理

78, 2025-03-01:
确实，人脸识别已经超出了esp32能力的范围，可以考虑其他解决方案

tju-wang, 2025-03-03:
我看sparkbot  esp32-s3有人脸识别的case，理论上是支持的

tju-wang, 2025-03-03:
你好邮件已收到、祝生活愉快！by tju_MrWang

tju-wang, 2025-03-03:
hi 虾哥，esp32-s3 使用ESP-WHO库，可以实现 face detection 和face recognition，支持录入人脸，检测模式下返回face recognition结果，这个feature我觉得很不错，可以与小智的音色识别结合，创造更多好玩的场景

78, 2025-03-03:
> hi 虾哥，esp32-s3 使用ESP-WHO库，可以实现 face detection 和face recognition，支持录入人脸，检测模式下返回face recognition结果，这个feature我觉得很不错，可以与小智的音色识别结合，创造更多好玩的场景

那可以，可以用来代替语音唤醒

StevenWang30, 2025-04-14:
> hi 虾哥，esp32-s3 使用ESP-WHO库，可以实现 face detection 和face recognition，支持录入人脸，检测模式下返回face recognition结果，这个feature我觉得很不错，可以与小智的音色识别结合，创造更多好玩的场景

这个现在有代码开源嘛？

tju-wang, 2025-04-14:
> > hi 虾哥，esp32-s3 使用ESP-WHO库，可以实现 face detection 和face recognition，支持录入人脸，检测模式下返回face recognition结果，这个feature我觉得很不错，可以与小智的音色识别结合，创造更多好玩的场景
> 
> 这个现在有代码开源嘛？

还没做，计划中；目前搭了开源小智服务器接入豆包，实现了图像内容理解