78/xiaozhi-esp32 #248: 立创·实战派ESP32-S3开发板使用小智v1.3.0固件唤醒的问题

Ariken2002, 2025-02-28:
立创·实战派ESP32-S3开发板使用小智v1.3.0固件，时不时会遇到多次才能唤醒的问题，如果设备刚刚从唤醒到待命，再马上唤醒，唤醒成功的几率要高一些，如果等了一段时间再唤醒，很多时候需要叫很多次才能唤醒，联系了立创官方的技术人员，他们让我把板子刷1.1的版本，看看有没有问题，我刷了1.1版本，测试发现还是有相同的问题，就是很多时候一次唤醒不了，要叫3次甚至更多次才能唤醒，不知道这是板子本身的问题，还是是小智固件的bug呢？麻烦虾哥给回复一下，感谢。

78, 2025-02-28:
你可以在控制台查看历史对话，听一下唤醒词的音频，看看有没有问题。
此前有报告过ZTS6216无法收音的情况。

findstr, 2025-03-01:
我也碰到了，甚至有时刚启动就不行，但是按一下按钮对话一下，能抗一点时间，待机一会又不行了

Ariken2002, 2025-03-01:
> 你可以在控制台查看历史对话，听一下唤醒词的音频，看看有没有问题。 此前有报告过ZTS6216无法收音的情况。

我在小智后台看了历史对话，在开发板无法唤醒没反应的时候，历史会话里也没有消息，只有唤醒成功的情况下，历史对话里才有那条唤醒的“你好小智”，看上去就像根本没有听到之前的唤醒词一样。

Ariken2002, 2025-03-01:
> 我也碰到了，甚至有时刚启动就不行，但是按一下按钮对话一下，能抗一点时间，待机一会又不行了

对的，跟我的情况差不多，而且发生得很频繁，经常遇到唤醒不了的情况，有时候甚至叫很多声都没反应。

Ariken2002, 2025-03-01:
目前立创开发板技术那边也不清楚到底是板子硬件问题，还是小智固件的原因，他们说他们对小智的固件也不大了解。

Ariken2002, 2025-03-01:
> 你可以试试注释 Application.cc 的 OutputAudio 里的以下代码：
> 
> ```
>             if (duration > max_silence_seconds) {
>                 codec->EnableOutput(false);
>             }
> ```
> 
> 看看问题是否仍然存在。

好的，我明天把这里改了编译一下试试看，多谢虾哥！

Ariken2002, 2025-03-03:
最近发现有可能是口音发音问题，我换了一个唤醒词，遇到相同情况的概率明显低很多了。

closed 2025-03-03 by Ariken2002