78/xiaozhi-esp32 #250: 立创·实战派ESP32-S3开发板,语音关闭后，机器无法开启

wushu-free, 2025-02-28:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v 1.3

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

在正常状态呀，我说“关闭”，这个命令词后，开发板就熄屏关闭了。这个时候，不管再怎么插拔电源，开发板一直都是黑屏状态，不管怎么按钮或者语音唤醒，都唤醒不了，也不亮屏，屏幕不显示任何东西

### What is the actual behavior?

在正常状态呀，我说“关闭”，这个命令词后，开发板就熄屏关闭了。这个时候，不管再怎么插拔电源，开发板一直都是黑屏状态，不管怎么按钮或者语音唤醒，都唤醒不了，也不亮屏，屏幕不显示任何东西

### Steps to reproduce.

对小智说关键词“关闭”

### Debug Logs.

```plain

```

wushu-free, 2025-02-28:
需要重新刷机后，再次联网，才会再次正常工作

zhuangxuan, 2025-02-28:
我的也是立创s3,没发生你说的情况，我用的智能体是英语老师

78, 2025-03-01:
怀疑它把屏幕亮度调为0了，你跟它说我屏幕看不见了，它应该会帮你调亮屏幕

wushu-free, 2025-03-02:
> 我的也是立创s3,没发生你说的情况，我用的智能体是英语老师

我的是实战派那个，我对他说了“关闭”，然后他就关屏了，然后就启不起来，再次插拔电源也没用，也唤不醒