78/xiaozhi-esp32 #255: ESP32面包板BOOT按键无法进入聆听模式

intcswalf, 2025-03-02:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

main V1.3.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

按下BOOT（IO0）按键，可以和小智持续对话。

### What is the actual behavior?

1、main分支下载到本地编译后，BOOT按键无法进入聆听模式，也就是说不能与小智对话。
2、修改config，把BOOT按键的配置改成其他引脚，按下按键可以和小智对话；
3、使用release的esp32面包板固件BOOT按键就可以正常打开。

### Steps to reproduce.

1、main分支下载到本地编译后，BOOT按键无法进入聆听模式，也就是说不能与小智对话。
2、修改config，把BOOT按键的配置改成其他引脚，按下按键可以和小智对话；
3、使用release的esp32面包板固件BOOT按键就可以正常打开。

### Build or installation Logs.

```plain

```

78, 2025-03-02:
@ooxxU 这个是不是要问一下你

ooxxU, 2025-03-02:
> [@ooxxU](https://github.com/ooxxU) 这个是不是要问一下你

理论上，1和3的应该是同样现象，但是不一样，我也有点不大清楚，开发者是什么开发板？或者进交流群沟通下，比较快捷。

intcswalf, 2025-03-08:
感谢回复，工作日比较忙没有上来看。
我用的是安信可的NODEMCU-32，模块是ESP32-S，芯片是ESP32。
已经进群了。

closed 2025-03-21 by 78