78/xiaozhi-esp32 #258: 怎样开启降噪

FenghenHome, 2025-03-03:
### Is your feature request related to a problem?

怎样开启降噪，我看idf.py menuconfig->ESP Speech Recognition，里面有个降噪

78, 2025-03-03:
如果用S3的芯片，默认应该是开启了的

FenghenHome, 2025-03-03:
afe_config_t afe_config = {
        .aec_init = false,
        .se_init = true,
        .vad_init = false,
        .wakenet_init = false,
        .voice_communication_init = true,
        .voice_communication_agc_init = true,
        .voice_communication_agc_gain = 10,
        .vad_mode = VAD_MODE_3,
        .wakenet_model_name = NULL,
        .wakenet_model_name_2 = NULL,
        .wakenet_mode = DET_MODE_90,
        .afe_mode = SR_MODE_HIGH_PERF,
        .afe_perferred_core = 1,
        .afe_perferred_priority = 1,
        .afe_ringbuf_size = 50,
        .memory_alloc_mode = AFE_MEMORY_ALLOC_MORE_PSRAM,
        .afe_linear_gain = 1.0,
        .agc_mode = AFE_MN_PEAK_AGC_MODE_2,
        .pcm_config = {
            .total_ch_num = channels_,
            .mic_num = channels_ - ref_num,
            .ref_num = ref_num,
            .sample_rate = 16000,
        },
        .debug_init = false,
        .debug_hook = {{ AFE_DEBUG_HOOK_MASE_TASK_IN, NULL }, { AFE_DEBUG_HOOK_FETCH_TASK_IN, NULL }},
        .afe_ns_mode = NS_MODE_SSP,
        .afe_ns_model_name = NULL,
        .fixed_first_channel = true,
    };
这里的afe_ns_mode 和afe_ns_model_name 吗，目前带的降噪算法是NSNET2吗，能否配置成NS_MODE_NET？

78, 2025-03-03:
设备上已经没有性能跑个降噪模型了

closed 2025-03-21 by 78

zhubinsheng, 2025-09-02:
> 设备上已经没有性能跑个降噪模型了

确认吗

zhubinsheng, 2025-09-03:
> > > 设备上已经没有性能跑个降噪模型了
> > 
> > 
> > 确认吗
> 
> 后面测试过 NSNET2，和WebRTC算法没有太大区别。你可以试试

没明白，使用S3自带的dsp跑降噪和aec不可以是把，此项目默认只有ace，但是降噪的话跑不动了是吧

zhubinsheng, 2025-09-03:
那可否加个外置的音频模块呢，那上面跑aec降噪等，同时也支持功放等，有推荐的芯片吗