78/xiaozhi-esp32 #265: 小智在说话的时候不能随时被打断

zptommy, 2025-03-04:
### Is your feature request related to a problem?

小智在说话的时候，希望能随时被打断，然后听我说话，这样体验会更好
否则小智说话很长的时候，就必须要等他说完，比较难受。

xuelse, 2025-03-05:
按下BOOT键或者说你好小智，也可以直接打断吧

sunq945, 2025-03-05:
是啊，按下BOOT键直接打断就好了！它毕竟不是人！如果一有声音就随时被打断的话，那么在人多的时候，或者环境声音比较嘈杂的时候，它就一直会被打断、听声，不能回应！还是有需要时按按键比较合理。

78, 2025-03-05:
> 是啊，按下BOOT键直接打断就好了！它毕竟不是人！如果一有声音就随时被打断的话，那么在人多的时候，或者环境声音比较嘈杂的时候，它就一直会被打断、听声，不能回应！还是有需要时按按键比较合理。

现在不就是这样用的吗

sunq945, 2025-03-08:
> > 是啊，按下BOOT键直接打断就好了！它毕竟不是人！如果一有声音就随时被打断的话，那么在人多的时候，或者环境声音比较嘈杂的时候，它就一直会被打断、听声，不能回应！还是有需要时按按键比较合理。
> 
> 现在不就是这样用的吗

是啊，我觉得这样就可以了

78, 2025-03-08:
也可以说话打断，但是面包板做不了AEC，需要有回采线路的硬件。群里已经有网友实现，后面可以合并到主线分支。

hyansuper, 2025-03-22:
> 按下BOOT键或者说你好小智，也可以直接打断吧

请问唤醒词能打断吗？有一个什么 duplex 配置有关系吗

closed 2025-04-01 by 78

luckyxiangfeng, 2025-05-04:
喇叭声音大声的时候，打断很难打断，要靠近麦克风才能打断