78/xiaozhi-esp32 #271: 设备拍照图片上传的协议更新

xinnan-tech, 2025-03-04:
### Is your feature request related to a problem?

有群友期望具有摄像能力的设备，拍照后，可将数据传输至服务器。希望虾哥团队新增一下后端接收图片的[接口协议](https://ccnphfhqs21z.feishu.cn/wiki/M0XiwldO9iJwHikpXD5cEx71nKh)。

### Describe the solution you'd like.

他提出，前端设备拍照后，按此协议进行传输，是否可行？
```
{
"type": "img",
"base64":"base64编码"
```

78, 2025-03-04:
图片是什么格式，有mime吗，设备上编码成jpeg大概是多少的体积，这个走mqtt还是websocket，可能需要拉个讨论群组针对这个进行设计。

peterpanstechland, 2025-03-07:
插眼

CrisXie4, 2025-03-07:
同求！

LanceCuby, 2025-03-08:
好主意，让有摄像头的设备支持拍照AI识图。协议可以参考下那些识图大模型的接口设计，融入到现在协议中。

666666g, 2025-03-08:
同求！

celesteStars, 2025-03-09:
有群踢我

666666g, 2025-03-09:
[image]我已经有了初步实现，但是被触发方式问题所困扰，有没有感兴趣的大佬来支援一下，vx：my46532

666666g, 2025-03-09:
[image]

vonweller, 2025-03-09:
> > [image]我已经有了初步实现，但是被触发方式问题所困扰，有没有感兴趣的大佬来支援一下，vx：my46532
> 
> 这周会先扩展IoT主动触发对话的协议。

目前已经适配所有支持视觉能力的llm，触发方式可以是iot也可以是拓展的协议
开源python地址
https://github.com/vonweller/PyxiaozhiVL

fliaping, 2025-04-03:
单图片这么做可以，还有视频流场景，模型参考Qwen2.5-Omni

78, 2025-04-04:
> 单图片这么做可以，还有视频流场景，模型参考Qwen2.5-Omni

适合建立 WebRTC 通道。

Gitsifu, 2025-04-11:
> 单图片这么做可以，还有视频流场景，模型参考Qwen2.5-Omni

允许接入全模态模型这个想法非常的酷！

dissipator, 2025-04-15:
好主意呢！同求

tju-wang, 2025-04-16:
> 好主意呢！同求 https://github.com/XuSenfeng/xiaozhi-alarm/tree/touch 这个有lichuang 的固件的，能弄个sparkbot的就更好了。我不会移植

doing

dissipator, 2025-04-21:
期待中

Oreofilling, 2025-04-22:
插眼

dissipator, 2025-05-05:
> > 好主意呢！同求 https://github.com/XuSenfeng/xiaozhi-alarm/tree/touch 这个有lichuang 的固件的，能弄个sparkbot的就更好了。我不会移植
> 
> doing

怎么样，我尝试自家动手，发现要么提示驱动问题，要么提示 
```
E (431) i2c.master: this port has not been initialized, please initialize it first
E (431) sccb-ng: failed to get SCCB I2C Bus handle for port 1
I (431) camera: Detected camera at address=0x03
E (441) camera: Detected camera not supported.
E (441) camera: Camera probe failed with error 0x106(ESP_ERR_NOT_SUPPORTED)
E (451) gdma: gdma_disconnect(319): no peripheral is connected to the channel
E (451) Camera: Camera init failed with error 0x106
```
@78  你好，能指导一下吗？
IDF-v5.4.1，espressif/esp32-camera: "^2.0.15"