78/xiaozhi-esp32 #285: 关于虾哥开源小智AI 升级大屏及数字人实时交互系统

Irwin0220, 2025-03-07:
我想把某个人的声音 样貌 保留在身边 只因为她不在了 我会私有化部署小智AI 并接入我沉积了1年的记忆库 如若成功 那就将开源归还予开源

Irwin0220, 2025-03-07:
是的 希望有志同道合的朋友 一起实现

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2025年03月07日 14:43 |
| 收件人 | 78/xiaozhi-esp32 ***@***.***> |
| 抄送人 | Irwin ***@***.***>,
Author ***@***.***> |
| 主题 | Re: [78/xiaozhi-esp32] 关于虾哥开源小智AI 升级大屏及数字人实时交互系统 (Issue #285) |

这是有个好想法，简单说就是做一个本地虚拟数字人，实现起来可能需要本地AI算力支持了吧

—

这是有个好想法，简单说就是做一个本地虚拟数字人，实现起来可能需要本地AI算力支持了吧

429367273, 2025-03-07:
支持

Irwin0220, 2025-03-07:
感谢

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2025年03月07日 16:38 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [78/xiaozhi-esp32] 关于虾哥开源小智AI 升级大屏及数字人实时交互系统 (Issue #285) |

支持

—

支持

achenakid, 2025-03-09:
支持，这个好有意义！

Irwin0220, 2025-03-09:
谢谢

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2025年03月09日 09:58 |
| 收件人 | 78/xiaozhi-esp32 ***@***.***> |
| 抄送人 | Irwin ***@***.***>,
Author ***@***.***> |
| 主题 | Re: [78/xiaozhi-esp32] 关于虾哥开源小智AI 升级大屏及数字人实时交互系统 (Issue #285) |

支持，这个好有意义！

—

支持，这个好有意义！

vonweller, 2025-03-09:
我这边已经在做基于小智ai的 unity数字人项目
目前已经完成能力对接。
在自定义的server32服务器上测试全流程都通过了，可惜在替换到小智官方的服务端时候出现了莫名的问题。

Irwin0220, 2025-03-09:
后端建议自己写一个

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2025年03月09日 18:50 |
| 收件人 | 78/xiaozhi-esp32 ***@***.***> |
| 抄送人 | Irwin ***@***.***>,
Author ***@***.***> |
| 主题 | Re: [78/xiaozhi-esp32] 关于虾哥开源小智AI 升级大屏及数字人实时交互系统 (Issue #285) |

我这边已经在做基于小智ai的 unity数字人项目
目前已经完成能力对接。
在自定义的server32服务器上测试全流程都通过了，可惜在替换到小智官方的服务端时候出现了莫名的问题。

—

我这边已经在做基于小智ai的 unity数字人项目
目前已经完成能力对接。
在自定义的server32服务器上测试全流程都通过了，可惜在替换到小智官方的服务端时候出现了莫名的问题。

vonweller, 2025-03-09:
> 后端建议自己写一个
> 
> ---- 回复的原邮件 ----
> | 发件人 | ***@***.***> |
> | 发送日期 | 2025年03月09日 18:50 |
> | 收件人 | 78/xiaozhi-esp32 ***@***.***> |
> | 抄送人 | Irwin ***@***.***>,
> Author ***@***.***> |
> | 主题 | Re: [78/xiaozhi-esp32] 关于虾哥开源小智AI 升级大屏及数字人实时交互系统 (Issue #285) |
> 
> 我这边已经在做基于小智ai的 unity数字人项目
> 目前已经完成能力对接。
> 在自定义的server32服务器上测试全流程都通过了，可惜在替换到小智官方的服务端时候出现了莫名的问题。
> 
> 
> 
> 我这边已经在做基于小智ai的 unity数字人项目
> 目前已经完成能力对接。
> 在自定义的server32服务器上测试全流程都通过了，可惜在替换到小智官方的服务端时候出现了莫名的问题。

后端对接好了自定义服务端的了已经全流程跑通啦，目前在对接3d模型口型与表情。

Irwin0220, 2025-04-23:
谢谢

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2025年04月23日 22:34 |
| 收件人 | 78/xiaozhi-esp32 ***@***.***> |
| 抄送人 | Irwin ***@***.***>,
Author ***@***.***> |
| 主题 | Re: [78/xiaozhi-esp32] 关于虾哥开源小智AI 升级大屏及数字人实时交互系统 (Issue #285) |
wx-11 left a comment (78/xiaozhi-esp32#285)

将生命永恒吗 也许是陪伴永恒 祝你成功

Hxert, 2025-05-02:
正在做这方面的工作，如能顺利完成再向你分享

Irwin0220, 2025-05-05:
可以加个联系方式 一起交流

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2025年05月03日 05:00 |
| 收件人 | 78/xiaozhi-esp32 ***@***.***> |
| 抄送人 | Irwin ***@***.***>,
Author ***@***.***> |
| 主题 | Re: [78/xiaozhi-esp32] 关于虾哥开源小智AI 升级大屏及数字人实时交互系统 (Issue #285) |
Hxert left a comment (78/xiaozhi-esp32#285)

正在做这方面的工作，如能顺利完成再向你分享