78/xiaozhi-esp32 #290: 新版本vad检测失效

randoms, 2025-03-07:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.4.4

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

之前版本在用户说话的时候红灯会高亮。

### What is the actual behavior?

用户说话时vad检测不再生效，板子上的指示灯颜色不再变换。

### Steps to reproduce.

刷入最新的1.4.4固件

### Debug Logs.

```plain

```

### More Information.

在这个 [commit](https://github.com/78/xiaozhi-esp32/commit/f1de536110ca42767745335a3984af5f3e0b5880) 中禁止了listen状态下的wakeup words检测。导致用户说话时vad检测不再生效，板子上的指示灯颜色不再变换。
还有请教一下为什么不使用esp32的vad，检测到有语音的时候在上传服务器，这样应该能大幅降低服务器的压力。

78, 2025-03-07:
vad也是大模型的一部分，小模型的效果可能不太理想。不同的模型性能不一样，如果有一个100KB的vad模型和一个2MB的vad模型，可能很多人也像试试2MB的模型的效果。

看来只能恢复listen状态下的检测，或者考虑在 audio_processor中打开vad。

randoms, 2025-03-08:
谢谢反馈，也感谢开源这么好的项目。

78, 2025-03-08:
指示灯不高亮的已经修复拉

closed 2025-03-10 by randoms