78/xiaozhi-esp32 #294: 能否增加对esp32-p4的支持

hbs2001, 2025-03-07:
### Is your feature request related to a problem?

希望可以添加对esp32-p4的支持，实现对更大屏幕更高刷新率的实现。

78, 2025-03-07:
之前 @Kevincoooool 好像实现过，p4好像还不是很流行？

Kevincoooool, 2025-03-07:
P4版本需要改动较多，待芯片量产上市后再添加即可，如果需要用P4版本可以先参考我的链接：https://github.com/Kevincoooool/xiaozhi-esp32/tree/FL7703_P4

hbs2001, 2025-03-07:
> P4版本需要改动较多，待芯片量产上市后再添加即可，如果需要用P4版本可以先参考我的链接：https://github.com/Kevincoooool/xiaozhi-esp32/tree/FL7703_P4

感谢大佬，我目前使用的开发板是ESP32-P4-Function-EV-Board，等下我尝试一下。

hbs2001, 2025-03-11:
> P4版本需要改动较多，待芯片量产上市后再添加即可，如果需要用P4版本可以先参考我的链接：https://github.com/Kevincoooool/xiaozhi-esp32/tree/FL7703_P4

这两天尝试了一下，始终无法编译通过，开始是代码中使用了未定义的配置宏 CONFIG_WIFI_RMT_CACHE_TX_BUFFER_NUM。改成DYNAMIC之后又报了许多来自managed_components/78__esp-wifi-connect/wifi_configuration_ap.cc的报错，麻烦大佬看一下是不是有什么改动没有合入。我使用的平台是win10 vscode，idf版本5.3.2。另外也尝试了5.4.0，同样编译不过但是报错不一样。

hbs2001, 2025-03-12:
@Kevincoooool 可不可以帮我看一下，麻烦了。

hbs2001, 2025-03-13:
是乐鑫esp-wifi-remote的bug，可以切换一下版本临时解决。
[https://github.com/espressif/esp-hosted-mcu/issues/39](url)

Alex-0x7C00, 2025-04-02:
添加了ESP32-P4-Function-EV-Board的支持

hbs2001, 2025-04-02:
SAMPLE_RATE改到24000之后正常了，感谢 @Alex-0x7C00

closed 2025-04-02 by hbs2001

Alex-0x7C00, 2025-04-02:
> SAMPLE_RATE改到24000之后正常了，感谢 [@Alex-0x7C00](https://github.com/Alex-0x7C00)

好的，我已经修改采样率到24000了

zhao2002-hash, 2025-06-11:
大佬，为什么开发板开启微信聊天界面风格之后没聊几句就会报 E (44511) Ml307AtModem: FIFO overflow错误。我的开发板是ESP32-p4 Function_EV_Board。我使用的是ML307 4G模块

78, 2025-06-11:
```
# Fix ML307 FIFO Overflow
CONFIG_UART_ISR_IN_IRAM=y
```
在 sdkconfig 里确保这个选项是开启的。

zhao2002-hash, 2025-06-11:
大佬，这个选项开启了但是依然会报这个错误

zhao2002-hash, 2025-06-12:
> 检查是否有任务优先级比 ML307 的优先级高。 试试在 `Application::OnClockTimer` 的开头里加入 `vTaskPrioritySet(NULL, 1);`

我在定时器函数里面开头加入这个函数，但是还是没有效果。另外由于我还没有摸清楚这个小智代码的逻辑，所以我不知道有没有任务的优先级比ML307高。

78, 2025-06-12:
去掉 `Application::OnClockTimer` 下面这行注释，你就能看到控制台每10秒打印的任务列表，就会显示优先级。
```
// SystemInfo::PrintTaskList();
```

zhao2002-hash, 2025-06-12:
`W (58290) MQTT: Received audio packet with wrong sequence: 127, expected: 126
W (58390) MQTT: Received audio packet with old sequence: 126, expected: 128
I (63669) Application: << 我男票上次试做结果炒成黑炭饭笑死！
| Task | Run Time | Percentage
| audio_detection  |   124787 |    6%
| esp_timer        |      923 |    0%
| IDLE0            |   881238 |   44%
| IDLE1            |   846042 |   42%
| sdio_read        |      170 |    0%
| sdio_process_rx  |      110 |    0%
| taskLVGL         |    13520 |    0%
| background_task  |    86299 |    4%
| main_loop        |    26792 |    1%
| sdio_write       |      940 |    0%
| rpc_rx           |        0 |    0%
| rpc_tx           |        0 |    0%
| sys_evt          |        0 |    0%
| ipc1             |        0 |    0%
| audio_communica  |        0 |    0%
| modem_event      |     1296 |    0%
| modem_receive    |    16101 |    0%
| Tmr Svc          |     2921 |    0%
| ipc0             |        0 |    0%
| sdio_rx_buf      |        0 |    0%
I (64152) Application: Free internal: 157787 minimal internal: 121447
I (67988) Application: << 要加猪油才香，不过现在流行用橄榄油啦～
I (72520) Application: STATE: listening
I (76694) Application: >> 什么是橄榄油？
I (76724) Application: STATE: speaking
I (77227) Application: << 就是地中海那边用橄榄榨的油啊！
E (77540) Ml307AtModem: FIFO overflow
| Task | Run Time | Percentage
| esp_timer        |      805 |    0%
| IDLE1            |   884543 |   44%
| IDLE0            |   975515 |   48%
| sdio_read        |      159 |    0%
| sdio_process_rx  |      109 |    0%
| taskLVGL         |      147 |    0%
| main_loop        |    20137 |    1%
| sdio_write       |      859 |    0%
| sys_evt          |        0 |    0%
| ipc1             |        0 |    0%
| ipc0             |        0 |    0%
| sdio_rx_buf      |        0 |    0%
| audio_communica  |        0 |    0%
| modem_receive    |        0 |    0%
| modem_event      |        0 |    0%
| background_task  |        0 |    0%
| audio_detection  |   115243 |    5%
| Tmr Svc          |     2636 |    0%
| rpc_rx           |        0 |    0%
| rpc_tx           |        0 |    0%
I (87152) Application: Free internal: 158895 minimal internal: 121447`

78, 2025-06-13:
是因为不是最新的源代码。

78, 2025-06-24:
> 没有啊，就是用了官方的7英寸的触摸屏，然后开启了微信聊天界面。然后启用ML307 4G模组。然后没了。

电平转换用的什么方案？晶体管电平转换电路不适用于波特率超过 460 kbps