78/xiaozhi-esp32 #296: 电量低弹窗一直不消失

127876, 2025-03-07:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

1.4.4

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

电量低显示几秒就可以了

### What is the actual behavior?

弹窗不消失

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

LFNL-scholar, 2025-03-07:
是不是刷错版本了

127876, 2025-03-07:
> 是不是刷错版本了

没有刷错，v1.4.4_xingzhi-cube-0.96oled-wifi，我是加了电池，电量是充满的，但还是显示电量低，而且这个弹窗一直不消失。

78, 2025-03-07:
> > 是不是刷错版本了
> 
> 没有刷错，v1.4.4_xingzhi-cube-0.96oled-wifi，我是加了电池，电量是充满的，但还是显示电量低，而且这个弹窗一直不消失。

右上角的电量图标，是空的吗？

127876, 2025-03-07:
> > > 是不是刷错版本了
> > 
> > 
> > 没有刷错，v1.4.4_xingzhi-cube-0.96oled-wifi，我是加了电池，电量是充满的，但还是显示电量低，而且这个弹窗一直不消失。
> 
> 右上角的电量图标，是空的吗？

对，是空的

78, 2025-03-07:
不会是有硬件问题吧，充满点，按理说不应该是空的。 除非检测不到电压呢？

xuelse, 2025-03-07:
> > 是不是刷错版本了
> 
> 没有刷错，v1.4.4_xingzhi-cube-0.96oled-wifi，我是加了电池，电量是充满的，但还是显示电量低，而且这个弹窗一直不消失。

你要使用这套硬件才能刷这个固件，这个固件算是针对它硬件专门做配置的

nicrowang, 2025-03-08:
display.cc 第155行可以修改屏蔽低电量弹框提示，可以修改为下面的，
        if (low_battery_popup_ != nullptr) {
            if (strcmp(icon, FONT_AWESOME_BATTERY_EMPTY) == 0) {
                if (lv_obj_has_flag(low_battery_popup_, LV_OBJ_FLAG_HIDDEN)) {
                    lv_obj_add_flag(low_battery_popup_, LV_OBJ_FLAG_HIDDEN);
                    // lv_obj_clear_flag(low_battery_popup_, LV_OBJ_FLAG_HIDDEN);
                    // auto& app = Application::GetInstance();
                    // app.PlaySound(Lang::Sounds::P3_LOW_BATTERY);
                }
            } else {
                // Hide the low battery popup when the battery is not empty
                if (!lv_obj_has_flag(low_battery_popup_, LV_OBJ_FLAG_HIDDEN)) {
                    lv_obj_add_flag(low_battery_popup_, LV_OBJ_FLAG_HIDDEN);
                }
            }
        }

YMQMKK, 2025-03-20:
我也遇到了相同的问题。[电量过低](https://github.com/xinnan-tech/xiaozhi-esp32-server/issues/437)

78, 2025-03-20:
> 我也遇到了相同的问题。[电量过低](https://github.com/xinnan-tech/xiaozhi-esp32-server/issues/437)

哪个板子，硬件是否已经支持电量检测？

YMQMKK, 2025-03-20:
> > 我也遇到了相同的问题。[电量过低](https://github.com/xinnan-tech/xiaozhi-esp32-server/issues/437)
> 
> 哪个板子，硬件是否已经支持电量检测？
板子：
[image]
现象：
已经充电很久了，却一直提示电量过低，请充电。
board type：
无名科技星智1.54(ML307)
商家提供的原理图：
[image]
[image]
通过跟踪 发现应该是在该函数中读取的adc值：
[image]
板子设计框架图：
[image]

78, 2025-03-20:
@WMnologo 看看是不是版本不一样

ywwt-net, 2025-07-02:
G17 引脚 直接3V3就没有弹窗了。