78/xiaozhi-esp32 #312: If someone creates a Display, they should call SetupUI themselves.

darkautism, 2025-03-08:
### Is your feature request related to a problem?

In main/display/lcd_display.h:25, SetupUI is declared as a virtual type and is likely intended to be overridden to customize the look and feel. However, according to https://isocpp.org/wiki/faq/strange-inheritance#calling-virtuals-from-ctors, virtual methods are not called during object construction. This means the overridden method will never be executed in this context.

### Describe the solution you'd like.

If SetupUI is moved to the public section, each board will need to call it explicitly. On the other hand, if it remains protected, we must find an alternative method to achieve the desired functionality.

### Describe alternatives you've considered.

It's uncertain whether these changes would disrupt the original logic.

78, 2025-03-08:
That is indeed true.
The SetupUI function in the descendant classes has never been functional. At present, all the CustomDisplay classes set up their UI elements within the constructor functions. Consequently, it would be advisable to remove the virtual keyword from the method.

78, 2025-03-21:
Fix this by removing the virtual keyword from SetupUI().

closed 2025-03-21 by 78