78/xiaozhi-esp32 #317: iot指令有时候会重复收到

HonestQiao, 2025-03-09:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.4.6

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

iot模块中提供的调用方法：
[image]

实际使用中，偶发出现被连续调用的情况：
[image]

类似的问题，也在其他开发板的音量设置上遇到过，重复设置多次。

### What is the actual behavior?

只有在理解对应设置意图的时候，才进行调用。

### Steps to reproduce.

偶发情况

### Debug Logs.

```plain

```

### More Information.

1. v1.4.6版本代码或者固件
2. MQTT连接模式

HonestQiao, 2025-03-11:
Qwen 实时

78, 2025-03-11:
> Qwen 实时

触发的时候，User的问题是什么还记得吗？

HonestQiao, 2025-03-17:
[image]

翻对话历史记录，找到了一些案发现场的情况

closed 2025-05-20 by 78